raspberrypi/documentation #1683: Add detailed information about DSI port on Raspberry Pi 4B

SomeAB, 2020-09-23:
Hello I have read the current version of the following information pages on the technical details of the display capabilities of the Raspberry Pi 4B:

https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/README.md
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/README.md

However I'm unable to reliably answer the following questions:
- The 2 lane MIPS DIS port's supported maximum resolution. 
- Revision version of the MIPS DIS (assuming that this is not MIPS DIS 2) 

Please provide and update the same information, on those pages and here on github. 

I ran into this, after I found that most of the Raspberry Pi 4B compatible displays available in India are very low resolution (less than FHD). I'm thinking of using a display from a smartphone for my project. However it seems that the higher resolution displays use 4 or 8 lanes of MIPS DIS. 

There is some obscurity in what revision of MIPS DIS is being talked about in most cases. 

(Disclaimer: Not an expert on displays)

JamesH65, 2020-09-23:
The resolution will depend on what frame rate you want, colour depth etc. Each lane is 1Gb/s IIRC, so with 2 lanes you have 2Gb/s.  Simply multiple the x by y (values must include front and back porch numbers) by fps by bits per pixel and that number needs to be less than that 2Gb/s. So its not a simple "This is a maximum resolution you can have"

And note, DPI is not the same as DSI.

lurch, 2020-09-23:
Is using a non-official DSI display still as tricky as this (old) [forum thread](https://www.raspberrypi.org/forums/viewtopic.php?t=188908) suggests it is?

JamesH65, 2020-09-23:
Yes, they are a real paint to get working.

6by9, 2020-09-24:
Yes DSI panels aren't straightforward. It's being worked on, and will be documented once resolved.

2 lanes is sufficient for a bit over 720p60. 4 lanes are required for 1080p60, so only an option on the Compute Modules.

It's DSI DPHY v1.1 IIRC.

SomeAB, 2020-09-26:
@JamesH65 So basically Raspberry Pi just called its port as 'DPI' ... which in all practicality it is just 2-lane DSI?

PS: apologies for not stating that I was talking about 24bit color, but yeah basically my question was what are the different combinations possible (upper limit of the DSI card)

Is @6by9 correct about the DSI version? Can someone crosscheck and update the same on the website (This is an important FAQ). I did not see anyone being able to use 1080p60 over that DSI connection while going through google results. 

Question: Btw, what 

I guess the best option is to just use the HDMI port, as most do. If it is 'DSI DPHY v1.1' as @6by9 has stated, I will close this issue. Just need a confirmation from someone.

lurch, 2020-09-26:
Nope, DPI = Display Parallel Interface - can be used to provide VGA-like signals over the GPIO pins https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/README.md

DSI = Display Serial Interface - where the [Raspberry Pi Touch Display](https://www.raspberrypi.org/products/raspberry-pi-touch-display/) gets attached https://www.raspberrypi.org/documentation/hardware/display/

6by9, 2020-09-26:
DPI is parallel - it uses a GPIO line per bit of the colour, plus hsync, vsync, pixel clock, and data enable.

DSI is serial - it uses differential signalling using a clock lane and 1 or more data lanes.

> Question: Does someone know the reason for one of the two DSI ports being disabled on the RPi 4B? Just curious to know the thought behind it.

Not disabled, just not routed out, mainly due to space constraints on the PCB. You could enable it, but it'd just put the signals to a set of balls on the underside of the SoC that you can't access.
Same reasoning why only 2 of the 4 data lanes are routed to the display connector.

lurch, 2020-09-26:
> Not disabled, just not routed out, mainly due to space constraints on the PCB. You could enable it, but it'd just put the signals to a set of balls on the underside of the SoC that you can't access.
Same reasoning why only 2 of the 4 data lanes are routed to the display connector.

See e.g. https://elinux.org/RPi_BCM2835_Pinout which shows which of the balls underneath the BCM2835 are routed out on the original Raspberry Pi Model B. Note that none of the DSI0_* signals are connected, and neither are lanes 2 or 3 of the DSI1_* signals.  (This is just an example - the pinout of the BCM2711 is probably vastly different to the pinout of the BCM2835, but the same PCB space constraints will apply). As mentioned by @6by9 , the full set of DSI signals are available on the [Compute Modules](https://www.raspberrypi.org/documentation/hardware/computemodule/datasheet.md).

SomeAB, 2020-09-28:
I guess all that is needed is adding this relevant information to the rasberry pi docs on the main website. I hope someone does it soon. Marking this as solved for now.

closed 2020-09-28 by SomeAB