raspberrypi/documentation #1685: Securing Raspberry PI - sudo visudo command usage

Zaederx, 2020-09-24:
While reading the [Securing Raspberry Pi Documentation](https://www.raspberrypi.org/documentation/configuration/security.md)  I found that the command:

 ```bash
sudo visudo /etc/sudoers.d/010_pi-nopasswd
```
 requires "-f" option before specifying file location. So maybe it should be change in the documentation to say:

```bash
sudo visudo -f /etc/sudoers.d/010_pi-nopasswd
```
[image]
[image]

JamesH65, 2020-11-18:
Odd, works fine for me. Looking at the visudo docs, it looks like the current installed version does not need the -f.

```
-f sudoers, --file=sudoers
                 Specify an alternate sudoers file location, see below.  As of version 1.8.27, the sudoers path can be specified without using the -f option.
```

JamesH65, 2020-11-18:
Because this works on the latest OS, closing.

closed 2020-11-18 by JamesH65