raspberrypi/documentation #1686: mistake in bootmode/net.md

ajs124, 2020-09-27:
I was trying to set up network booting with a raspberry pi 4 and ran into some issues in the documentation.

Namely I'm not running `dnsmasq`, which the documentation always seems to assume.

E.g. [it specfies](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/net.md#dhcp-request--reply) that you need to set the vendor option 43 so that it 

> contain[s] the string "Raspberry Pi Boot"

Luckily there's a tcpdump next to it, otherwise I would have never figured out that you actually don't just need to send that string but surround it by a bunch of non-printable ASCII, because that is what `dnsmasq` does.

The whole PXE implementation seems kind of weird. Why does it send 0 (x86/BIOS) as architecture option? Why does it require this random string in option 43? Is the code for it open or closed?

lurch, 2020-09-28:
That page was originally written around the network-boot option included in the Pi 3B and Pi 3B+. This is embedded inside the bootloader in the BCM2837 SoC and is non-updateable, which explains why it has "quirks".

The Pi4 bootloader by comparison is stored in [EEPROM](https://github.com/raspberrypi/rpi-eeprom) so any glaring quirks can be fixed, but I guess it also needs to maintain backwards-compatibility with the Pi 3B+ network-boot mode?
@timg236 knows more about this than I do...

The bootloader code for both the Pi 3B+ and Pi 4B are closed-source.

timg236, 2020-09-28:
This has already been discussed here https://github.com/raspberrypi/rpi-eeprom/issues/182 and is unlikely to change

ajs124, 2020-09-28:
@timg236 I only see the architecture option being discussed there, not the contents of or reasoning for option 43.

dasJ, 2020-10-06:
In case anyone stumbles over this: You need this abomination in your dhcpd config:

```
# What the hell raspberrypi foundation
option option-43 code 43 = string;

if substring(hardware,1,3) = b8:27:eb or substring(hardware,1,3) = dc:a6:32 {
  next-server 10.66.3.101;
  option option-43 06:01:03:0a:04:00:50:58:45:09:14:00:00:11:52:61:73:70:62:65:72:72:79:20:50:69:20:42:6f:6f:74:ff;
}
```

dasJ, 2020-10-06:
Well I don't think anything will break when implementing the bootloader acording to the documentation. What dnsmasq sends does **contain** the specified string, there is no actual reason to test the rest of the ascii characters. Either documentation or (preferrably) the code should be fixed.

JamesH65, 2020-11-19:
Since the bootcode is unlikely to change, what docs changes would be appropriate and where?

ajs124, 2020-11-19:
Basically, I'd suggest to rephrase this whole paragraph. Keep the tcpdump output, give an example for dnsmasq and isc-dhcpd and just say something along the lines of "when in doubt, adhere to the tcpdump output" instead of stating verifiably wrong things, as it does now.

Anyways, I'll close this, because idgaf tbh.

closed 2020-11-19 by ajs124