raspberrypi/documentation #1696: [Pi 4B] Boot from 400 GB San Disk Micro SD Card?

PoisnFang, 2020-10-10:
I just want to verify. Can I use a [400 GB San Disk Micro SD Card](https://www.amazon.com/gp/product/B074RNRM2B) to boot from?

JamesH65, 2020-10-10:
This repository is for reporting issues with the documentation, please use the forums for technical support.

closed 2020-10-10 by JamesH65

PoisnFang, 2020-10-10:
@JamesH65  That's the thing... the documentation does specify...

JamesH65, 2020-10-10:
We do not have lists of cards that work - it would be impossible to maintain. So your only option is to ask on the forums if a specific device works. Note, 99.99% of cards do work.

PoisnFang, 2020-10-10:
Ok let me re-specify then:

>Note: Only the Raspberry Pi 3A+, 3B+ and Compute Module 3+ can boot from an SD card larger than 256 GB. This is because there was a bug in the SoC used on previous models of Pi.

Can a RPi4 B Boot from a card that is greater than 256 GB?

PoisnFang, 2020-10-10:
Specifically this mentions "previous models of Pi" so does that mean that a 400 GB will work with a Pi 4B since it is not a previous model? @JamesH65

JamesH65, 2020-10-10:
www.raspberrypi.org/forums

But yes, as far as I know, but I've never done it.

JamesH65, 2020-10-10:
Which page is that text on? I'll update it.

PoisnFang, 2020-10-10:
https://www.raspberrypi.org/documentation/installation/sd-cards.md