raspberrypi/documentation #1699: Multi-lang documentation

felixmallet, 2020-10-14:
Who are courage to help me to translate the documentation in other language (examples: french(I'm speak french lol), spanish, etc)

lurch, 2020-10-15:
We've had similar requests before ( #328 #335 #733 #1082 ), but unfortunately we don't have any systems in place for hosting translated versions of the documentation. (and we'd have no way of ensuring that the translated content is kept up-to-date with the original version.) :earth_africa: 

However the [licence](https://github.com/raspberrypi/documentation/blob/master/LICENCE.md) means you're free to fork the repo and host your own translated versions :slightly_smiling_face:

JamesH65, 2020-11-18:
Closing as a Won't Fix. This means that although the issue is acknowledged, it is felt that no further action is likely. This may be down to a bad benefit vs cost analysis, or it may actually not be possible to fix.

closed 2020-11-18 by JamesH65