raspberrypi/documentation #1704: Add Compute Module 4 informations

TimonPeng, 2020-10-23:
Like mechanical drawings...

aallan, 2020-10-23:
There are STEP files [for the CM4](https://datasheets.raspberrypi.org/cm4/CM4-step.zip) available for the CM4, and mechanical drawings can be found [in the datasheet](https://datasheets.raspberrypi.org/cm4/cm4-datasheet.pdf). Mechanical drawings for the CM4IO can be found [in the datasheet](https://datasheets.raspberrypi.org/cm4io/cm4io-datasheet.pdf), and the [KiCad design files](http://datasheets.raspberrypi.org/cm4io/CM4IO-KiCAD.zip) for the CM4IO are also available.

All of these are linked off the [CM4](https://www.raspberrypi.org/products/compute-module-4/) and [CM4IO](https://www.raspberrypi.org/products/compute-module-4-io-board/) Product pages.

lurch, 2020-10-23:
Maybe various places (or new pages) in the documentation repo should link to those files @aallan , since that's where those files have been located previously, e.g. 
https://www.raspberrypi.org/documentation/hardware/computemodule/schematics.md
https://www.raspberrypi.org/documentation/hardware/computemodule/datasheet.md
https://www.raspberrypi.org/documentation/hardware/raspberrypi/mechanical/README.md

aallan, 2020-10-23:
Yup, @JamesH65 is doing it now.

JamesH65, 2020-10-23:
Actually...now.

JamesH65, 2020-10-23:
https://github.com/raspberrypi/documentation/pull/1705

JamesH65, 2020-11-18:
Closing this issue as questions answered/issue resolved.

closed 2020-11-18 by JamesH65