raspberrypi/documentation #1709: Missing PiZero camera instructions

lurch, 2020-10-26:
Might be nice if https://github.com/raspberrypi/documentation/blob/master/usage/camera/installing.md also described how to connect a camera module to a PiZero?

andrum99, 2020-10-27:
Some things it should mention:

- Where the camera connector is located on the Pi Zero
- The fact that not all Pi Zeros have a camera connector
- How to open and close the camera connector on the Pi Zero
- The fact that the Pi Zero needs a different cable to connect the camera to the Pi Zero, since the connector on the Pi Zero is smaller
- Where to get the special camera cable for the Pi Zero

aallan, 2021-08-13:
The camera documentation is undergoing a total rewrite, and this has been flagged internally for that. Closing.

closed 2021-08-13 by aallan