raspberrypi/documentation #1712: CM4IO designed using pre-release version of KiCAD

craigarno, 2020-10-29:
I downloaded the CM4IO KiCAD files and discovered the files can't be opened using latest KiCAD production release version (5.1.7)-1.

Unfortunately it looks like this schematic/PCB design was created using a bleeding edge development "nightly" build which is -not- to be used for production.  The apparent nightly build version used for this design is from version 20200828.  Nightly version 20200828 is no longer available for download.

Using a current nightly kicad-r18494.e739d5ba65-x86_64.exe (v5.99 10/27/2020), the KiCAD PCB design is broken.  None of the PCB traces are visible.  There are also new PCB "Board Setup" data fields like "Differential Pairs" which are missing data needed by the current nightly KiCAD PCB builds.

It looks like a fair amount of work was done for this board design.  The Schematic part still seems to work.  Now to use the PCB design it needs to be redone or this SCH & PCB design package moved to 5.1.7 "production release" to be used.  Or wait for 6.0 to become a "Released-Stable" version to go in and correct all the problems created by using a bleeding edge KiCAD development release for "production" design.  5.1.7 designs should transition to 6.0 when 6.0 is released.

What to do in the meantime?  The PCB design without traces is largely not usable and back tracking to the nightly build which created this design is also not possible.
[image]
[image]

aallan, 2020-10-29:
> The apparent nightly build version used for this design is from version 20200828. Nightly version 20200828 is no longer available for download. Using a current nightly kicad-r18494.e739d5ba65-x86_64.exe (v5.99 10/27/2020), the KiCAD PCB design is broken.

Nightlies as recent as launch day, which was the 19th October, loaded the design files correctly.

JamesH65, 2020-10-29:
I think the problem is that the launch day version is no longer  available, and the latest nightly does not work. So effectively there is no available KiCAD that will read the files.

aallan, 2020-10-29:
Nightlies as far back as the 15th of October are available for download from the KiCad Downloads page, e.g. for Windows as the OP seems to be on Windows, see https://kicad-downloads.s3.cern.ch/index.html?prefix=windows/nightly/

dp111, 2020-10-29:
Works for me. You may not be know how the latest version of Kicad works. See the Kicad blog for more details : https://kicad-pcb.org/blog/2020/10/Development-Highlight-PCBNew-Layers-DRC-and-Toolbar-Palettes/

aallan, 2020-10-29:
The UI of KiCad has changed somewhat. See the comments on layer management in the [recent blog post](https://kicad-pcb.org/blog/2020/10/Development-Highlight-PCBNew-Layers-DRC-and-Toolbar-Palettes/). If you go to the Appearance Control panel and click on the Objects tab you can find the visibility control for traces there.

[image]

Jaronasdf, 2020-10-29:
What happened to the 3D models in the zip file? They are not included anymore.. Does anyone have an older version they can share that has the 3D models? I need it for designing a housing.

dp111, 2020-10-29:
I'm in the process of updating the 3dmodels to correct a few issues I've spotted I'll make a new release soon

craigarno, 2020-10-29:
That fixed it for me in the nightly I downloaded.  On the second tab [Objects]; Tracks, Vias, and Pads are "off" by default (sub-optimal default selections for a "new" feature).  I use the Stable version 5.1.7 for production work.  Nightlies are a moving target so not desirable unless I absolutely need a new 6.0 feature.  Your CM4IO design is usable now, Thank you.

If you figure out why some of the 3D parts, mostly connectors don't show, I'd like to know this too.  Otherwise my original issue is addressed.

[image]
[image]

closed 2020-10-30 by aallan

aallan, 2020-11-02:
> If you figure out why some of the 3D parts, mostly connectors don't show, I'd like to know this too. Otherwise my original issue is addressed.

See the comment above upthread about the 3d models. They have been temporarily removed.