raspberrypi/documentation #1717: cm4 kicad5 footprint

aep, 2020-11-02:
would be nice if there was an officially sanctioned footprint for kicad5.

JamesH65, 2020-11-02:
As I understand it, KiCAD5 cannot do some of the things we need which is why we use the development versions. So not sure suppling a v5 version is possible.

aallan, 2020-11-02:
This is correct, there is no way we can provide KiCad 5 versions of the files as there was a bug which surfaced when routing that needed a nightly build. We'll probably revise and release an official KiCad 6 version, but until that happens you'll need to use a development version of KiCad. Sorry.

aep, 2020-11-02:
Uuuh that sounds odd. The footprint isn't very complicated... I'll create the footprint then and see if the bug can't be worked around.

aep, 2020-11-02:
did not encounter any bugs making this footprint.

https://github.com/aep/kicad/blob/master/lib/aep3.pretty/rpiCM4.kicad_mod

would be appreciated if someone could review it for the benefits of others.

[image]

aallan, 2020-11-02:
Oh! Okay, sorry. I thought you were talking about the KiCad design files for the CM4IO board, as we haven't supplied any KiCad files for the CM4 Module itself. The CM4 was not designed in KiCad. 

However we did supply [STEP files](https://datasheets.raspberrypi.org/cm4/CM4-step.zip) for the CM4 module?

aallan, 2020-11-02:
> the official IO board files already dont open anymore on master from today

In relation to this part of your question, it's possible you've also run into the UI changes we say in #1712 that led someone else to think that the files weren't opening.

aep, 2020-11-02:
> However we did supply STEP files for the CM4 module?

yes, thank you. this is very helpful to verify the footprint

> In relation to this part of your question, it's possible you've also run into the UI changes we saw in #1712 that led someone else to think that the files were broken in the current nightly.

nah, got a bunch of unconnected NET errors. But its fine, i got a footprint now :)

dp111, 2020-11-02:
Can you give an example of one of the unconnected nets ?

aep, 2020-11-02:
did a reinstall of current master (unfortunately requires make install to switch versions) , cannot reproduce. looks fine *shrug*

closed 2020-11-02 by aep

garlick, 2020-11-19:
Nice!  Thanks so much for making this footprint.  This would be a nice one to have in the official kicad distribution, if you were willing to submit a PR.