raspberrypi/documentation #1728: Wireless Regulatory Domain [Country Code] in configuration/wireless/desktop.md

DeeJay, 2020-11-10:
configuration/wireless/desktop.md says -
"Note that on the Pi 3B+, which supports the 5G protocol, wireless networking is disabled for regulatory reasons, until the country code has been set. "

Presumably this is also true of later hardware, including RPi4B and RPi400 at the time of writing?

JamesH65, 2020-11-10:
I believe that is correct. I'll update the docs when I get the chance.

JamesH65, 2020-11-10:
@XECDesign Can you confirm?

aallan, 2020-11-10:
We should also change 5G to 5GHz, as I'm presuming we're talking about 5GHz wireless networking here, rather than 5G celluar?

lurch, 2020-11-10:
Is the country code also set by the startup-wizard @spl237 ? If it is, that's probably worth mentioning here too?

spl237, 2020-11-10:
Yes it is.

XECDesign, 2020-11-16:
> @XECDesign Can you confirm?

Yup, this applies to al 5GHz models... or should. I may be wrong, but IIRC, CM4 uses a different chip, so I'll need to check that this works before the next image release.

JamesH65, 2020-11-18:
Fixed on master.

closed 2020-11-18 by JamesH65