raspberrypi/documentation #1730: Link bitrot

JamesH65, 2020-11-12:
From info@raspberrypi.com

Link bitrot! On page https://www.raspberrypi.org/documentation/installation/sdxc_formatting.md there's: "A simple tool to do this is FAT32 Format which"

...which points to http://www.ridgecrop.demon.co.uk/guiformat.htm . Except that's wrong -- drop the www. and everything works. Of course their program ALSO references www. on it's final output screen, so maybe they should reinstate the www CNAME instead.

I of course hit this trying to FAT32 format a 64GB uSD card; the official native program did wonderful making an exFAT system, which is great but unusable here. You might say NOT to use that. You also might want to bold the last sentence of the 2nd paragraph, just because you can.

And a possible backup copy of that program is: https://download.cnet.com/FAT32format/3000-2094_4-77594227.html.

JamesH65, 2020-11-12:
I tried this and it seem to work in Chrome, but my be browser specific, so should probably be fixed anyway.

lurch, 2020-11-12:
When I try visiting http://www.ridgecrop.demon.co.uk/guiformat.htm it auto-redirects to http://www.ridgecrop.demon.co.uk/index.htm?guiformat.htm (which displays the content I'd expect).

And if I "drop the www" as the email suggests, i.e. http://ridgecrop.demon.co.uk/guiformat.htm then I get "server IP address could not be found".

Strange... (maybe demon.co.uk have been fiddling about with how they host customer websites? :man_shrugging: )

aallan, 2020-11-12:
I see the same as @lurch

JamesH65, 2020-11-18:
Hmm, all seems to work fine for everyone who has tried it, and the proposed fix definitely doesn't work, so closing.

closed 2020-11-18 by JamesH65