raspberrypi/documentation #1733: Cross-compilation kernel manual errors

baranchuk, 2020-11-15:
Try to make cross-compile kernel with tca8418 support. Make it on clean ubuntu 18.04
Have a error at step 
sudo make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=~/linux/mnt/ext4 modules_install
get error
ln: target 'support+/source' is not a directory
Makefile:1349: recipe for target '_modinst_' failed
make: *** [_modinst_] Error 1
still can compile at raspberry, but 0W take about 32-33hrs....

lurch, 2020-11-15:
If there _were_ errors in the documentation, I'm sure there'd have been other reports of them by now? :shrug: 

When comparing your command-line above to [the documentation](https://github.com/raspberrypi/documentation/blob/master/linux/kernel/building.md) it looks like you're missing the `env PATH=$PATH` part? (although I've no idea if that's what's causing your error). Also, are you sure your `INSTALL_MOD_PATH` is correct?

I suggest you ask for help on the [forums](https://www.raspberrypi.org/forums/). If you **do** manage to identify an error in the documentation, please feel free to submit a Pull Request with the fix.

closed 2020-11-15 by lurch

lurch, 2020-11-15:
P.S. If this driver is buildable as a module, you could try opening an issue at https://github.com/raspberrypi/linux/ asking for it to be included in the default RPi kernel config?

baranchuk, 2020-11-23:
I try to complile kernel by this manual
https://www.raspberrypi.org/documentation/linux/kernel/building.md
addind support tca8418 keyboard driver by this manual
https://www.raspberrypi.org/documentation/linux/kernel/configuring.md
and this link
https://5volt-junkie.net/i2c-keypad/?fbclid=IwAR3sxODrIp4wIauTaCgZ4s6OeQSRoyfbkbda13flbZfDTxKqNJhWSptvkDI

Try it on fresh Ubuntu 18.04, Debian 4.9.110 and directly on RPI0W with latest upgraded Raspian.
After command
sudo env PATH=$PATH make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=mnt/ext4 modules_install on Debian and Ubuntu or sudo make modules_install on Raspian I get the same error message

```
ln: target 'support+/source' is not a directory
Makefile:1349: recipe for target '_modinst_' failed
make: *** [_modinst_] Error 1
```
Any suggestions please.

lurch, 2020-11-23:
I've just tried this myself using the two links you posted
https://www.raspberrypi.org/documentation/linux/kernel/building.md
https://www.raspberrypi.org/documentation/linux/kernel/configuring.md
and it all worked absolutely fine on Ubuntu 18.04.5, and I end up with:
```
$ ls -l mnt/ext4/lib/modules/5.4.77+/kernel/drivers/input/keyboard/tca8418_keypad.ko 
-rw-r--r-- 1 root root 9260 Nov 23 02:30 mnt/ext4/lib/modules/5.4.77+/kernel/drivers/input/keyboard/tca8418_keypad.ko
$ file mnt/ext4/lib/modules/5.4.77+/kernel/drivers/input/keyboard/tca8418_keypad.ko 
mnt/ext4/lib/modules/5.4.77+/kernel/drivers/input/keyboard/tca8418_keypad.ko: ELF 32-bit LSB relocatable, ARM, EABI5 version 1 (SYSV), BuildID[sha1]=441cbcaf95912baecf8925eb5d21256fa7cafea3, not stripped
```
which shows that it's all worked okay (as I would expect it to).
So I'm afraid you must be doing something wrong, and as I said before you'll have to ask for help on the forums https://www.raspberrypi.org/forums/

lurch, 2020-11-23:
I just did a bit of poking about in the kernel `Makefile` (because I was curious), and it looks like things would break if you're using a space in your custom kernel-version, so make sure you're not doing that.

baranchuk, 2020-11-23:
Thank you so much, space is a problem!!!! How i can donate you some beers?