raspberrypi/documentation #1736: No Audio Output Selection possible

IsabellLehmann, 2020-11-17:
Hi, in the documentation (https://github.com/raspberrypi/documentation/blob/master/configuration/audio-config.md) it is explained how to set the Audio Output.

However, maybe there was an update, but I do not have this option on my raspberry Pi 4B. I installed the current version of Raspbian and also used sudo apt-get update and upgrade to really get the newest version.

This is how my menu looks:
[image]
and in the advanced options:
[image]
So there is no possiblity to change the audio output.

IsabellLehmann, 2020-11-17:
The documentation needs to be updated that the audio is now set in the System options:
[image]

lurch, 2020-11-17:
I agree that the documentation about `raspi-config` is probably outdated (see also e.g. #1041 ), but FYI you can also control most of these raspi-config options using the "Raspberry Pi Configuration" GUI program, which you can find in the "Preferences" menu on the desktop. Except the audio options, which you can configure by right-clicking on the Volume icon in the top-right corner of the screen :slightly_smiling_face:

JamesH65, 2020-11-19:
OK, have updated the page to remove the images (they are such a PITA to keep updated) and corrected the usage to reflect the current version of raspi-config.

closed 2020-11-19 by JamesH65