raspberrypi/documentation #1746: dts files for cm4 using cam and disp

jamwu, 2020-11-25:
Can you update the example dts files for configuring the camera and display on cm4?

JamesH65, 2020-11-25:
Can you provide a link to the page that is incorrect or needs updating please.

jamwu, 2020-11-25:
> Can you provide a link to the page that is incorrect or needs updating please.
here is the link.
https://github.com/raspberrypi/documentation/blob/master/hardware/computemodule/cmio-camera.md

kralo, 2020-12-08:
I am stuck as well. Cross-Linking Discussion from raspberry Forum: [https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=292280&p=1767256#p1767081](https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=292280&p=1767256#p1767081)

6by9, 2020-12-08:
I'm updating all the DTS / BIN and instructions at the moment.

6by9, 2020-12-08:
See #1755 for the blobs as a work in progress.

kralo, 2020-12-08:
I took some photos if that helps.
https://ibb.co/MNX8mqJ
https://ibb.co/q9Y6XPH
https://ibb.co/NF2mR0n

However, I was unable to verify the DTS's. I have tried to compile them as overlays and use with ov9281, but no luck. I'll wait for the written documentation and test again.

6by9, 2020-12-08:
The request for images was more for my colleagues than for general contributors. We have in-house photography/videography people who any image requests would normally get routed to.

They aren't overlays, and precompiled bin files are already provided. As per the existing docs, the appropriate one of those gets copied to /boot/dt-blob.bin.

dt-blob.bin configures the firmware only, not the Linux kernel.
ov9281 is not supported by firmware, therefore it won't do anything directly with it. ```vcgencmd get_camera``` will never report an ov9281 as detected.

The one thing that is passed from firmware to kernel is that the firmware will look for DT overrides called
```cam0-pwdn-ctrl```
```cam0-pwdn```
```cam0-led-ctrl```
```cam0-led```
and the same for cam1. If found then it will update them based on the camera GPIO configuration that is in dt-blob.bin.
The I2C pinmuxing is not passed through, so the kernel/ov9281 overlay will assume it is on i2c_csi_dsi (aka i2c-10) on probably.
Having just tried that though, the pinctrl configuration for the I2C lines on CM3 isn't correct (it muxes BSC0 to both 0&1 and 28&29 simultaneously), so the sensor isn't detected.

On a CM4 with an Inno-maker OV9281 module connected to CAM1, ```dtoverlay=ov9281``` with no /boot/dt-blob.bin works fine. If you need to connect anything differently then I suggest you fork the ov9281 overlay as dt-blob.bin is the wrong place to be configuring kernel drivers.

ansonhe97, 2020-12-15:
Hi @6by9 

I was testing the modified `dts` bin(also build from source`.dts`) and cannot get it working. I'm using Rpi camera v2.1 with CM4 on IO board, and connected to CAM1.

```sh
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
pi@raspberrypi:~ $ vcgencmd get_camera
supported=1 detected=0
```

Furthermore, would there be a `.dts` for enabling only CAM0 (or CAM0 + DISP0)? 

Any help would be much appreciated. Thanks in advanced!

Anson

6by9, 2020-12-15:
This isn't the place to be debugging your DTS files - please use the Compute Module or Camera Board sections of the forum.

I wasn't planning on creating files for just CAM0 or CAM0+DISP0. There are too many combinations to make it easily maintainable.
The documentation describes the changes required in the dt-blob.dts file, and all the files are now rebased so that a simple comparison of the files will show what has been changed in each case.

aallan, 2021-06-07:
Will be resolved by #1911.

closed 2021-06-07 by aallan