raspberrypi/documentation #1747: cm4: suggest to clarify RX/TX directions

aep, 2020-11-27:
the pcie RX/TX pins on the CM4 seem to be labeled from the perspective of the device rather than the host, so i incorrectly connected cm4 RX to device TX.  I'm not sure if this is just a noob mistake on my end, but it would be helpful if the datasheet contained a simplified schematic to prevent others from making that mistake

JamesH65, 2020-11-27:
@dp111 FYI

dp111, 2020-11-27:
A PCIe card that is compliant with the PCIe specification will have the the TX decoupling capacitor on board, so no need for them on the CM4IO board. These TX lines will connect to the RX lines on the CM4IO board.

aep, 2020-11-27:
thanks. is that the same case for nvme?

either way, can you clarify if PCIe_RX needs to be attached to device RX (as done on the io board) or device TX (as mentioned in the datashset)

aep, 2020-11-27:
that doesn't answer the question unfortunately. 

if RX is an input, it would have to be connected to the pcie device output, but the cm io doesnt do that.
i also see a clock signal on PCIe_RX.

dp111, 2020-11-27:
The PCIe connector is labelled on the CM4IO from the host point of view as is the convention . Is this what is confusing ?

aep, 2020-11-27:
yes! that is the confusion, thank you. i have never wired up pcie, so i didnt realize this is probably standard?

is it the same on nvme?

http://read.pudn.com/downloads794/doc/project/3133918/PCIe_M.2_Electromechanical_Spec_Rev1.0_Final_11012013_RS_Clean.pdf   page 155 says pin 49 is PETp0, so i just wire that to rpi PCIe_TX?

dp111, 2020-11-27:
I suspect is it is the same , but I don't have any of my nvme stuff to hand to check. 

From the CM4 datasheet point of view if you are wiring to a chip directly ( not using a socket ) then you do have to to TX to RX and add the capacitors on TX. With the external PCIe card this is taken care of. So its a bit complex.

dp111, 2020-11-27:
Looking at my nvme drive and adaptor is does look as though socket is labelled from the host point of view too.

aep, 2020-11-29:
thank you! nvme is coming up fine for me now. Unfortunately there's no boot from nvme option, but thats a different issue.

I still think the docs should contain a hint that the RX->TX swap doesn't apply to most use cases. Feel free to close this issue if you disagree.

dp111, 2020-11-30:
Depends on what you see as most cases. I have made some changes that will be in the next release of the datasheet

aallan, 2021-06-07:
> I have made some changes that will be in the next release of the datasheet

Closing as idle.

closed 2021-06-07 by aallan