raspberrypi/documentation #1748: documentation/configuration/wireless/headless.md contain an error

NielsGx, 2020-11-28:
Hello,

in the headless.md file, you should say that we need to copy the wpa_supplicant.conf file into the boot partition, instead of saying boot folder.
I thought I needed to mount the ext4 partition, and then copy the wpa file into the boot folder, which is clearly not the same location than in the small boot partition.

You could inspire yourself from documentation/remote-access/ssh/README.md :

"For headless setup, SSH can be enabled by placing a file named ssh, without any extension, onto the boot partition of the SD card from another computer. When the Pi boots, it looks for the ssh file. If it is found, SSH is enabled and the file is deleted.[...]"

and also this in the end :

"If you have loaded Raspberry Pi OS onto a blank SD card, you will have two partitions. The first one, which is the smaller one, is the boot partition. Place the file into this one."

lurch, 2020-11-29:
Does https://github.com/raspberrypi/documentation/commit/9c649c9602069c3973c7d741c15bf07152227132 help ? :slightly_smiling_face:

NielsGx, 2020-11-29:
No.

As I said, boot partition and boot folder is not the same thing. Should be boot partition, the only partition that is readable/writable natively by windows mac OS. Should be a fat32 partition, if you used the imager tool, at least on windows.

As I said, there's a boot FOLDER in the other partition, the "real" partition, in ext3/ext4. And it's not here we should put the files..

Thanks for your help.

NielsGx, 2020-11-29:
for a lite image, the fat32 boot partition is like 300mb, and the "real" ext4 partition is more than 1Gb.

So ssh and wpa_supplicant have to be out in the root of boot PARTITION. 

not in the real partition/boot folder.

NielsGx, 2020-11-29:
replacing the word folder by partition and adding this at the end should be enough imho.

"If you have loaded Raspberry Pi OS onto a blank SD card, you will have two partitions. The first one, which is the smaller one, is the boot partition. Place the file into this one."

aallan, 2021-08-09:
Source file now lives at computers/configuration/headless.adoc

aallan, 2021-08-13:
...and has been reworded.

closed 2021-08-13 by aallan