raspberrypi/documentation #1754: UART: Enabling of UART1 for CM3+ does not work

boraozgen, 2020-12-07:
Hi all,

I spent a couple of hours today trying to enable UART1 (Mini UART) for my CM3+. I tried the `enable_uart=1` and `core_freq=250` settings as depicted in the documentation to no avail. After reading some forum posts and github issues, the actual solution for me was to set `dtoverlay=uart1,txd1_pin=32,rxd1_pin=33` in `config.txt`. I'm not sure if this is CM specific, but I think it should be added to the documentation.

Cheers

lurch, 2020-12-07:
ping @pelwell

pelwell, 2020-12-07:
That's the correct behaviour. Rather than enable a lot of devices on the off-chance that customers want them, the CM DTS files opt to leave most optional devices disabled. It was always expected that customers would customise the DTS/DTB to suit their application, or provide a custom overlay incorporating the necessary changes.

boraozgen, 2020-12-07:
It was not clear to me that UART1 is one of the "changes" from the main board. Anyway, thanks for the clarification and you can close the issue if you see no need for additions.

pelwell, 2020-12-07:
UART1 is "changed" from the "main board" (I assume you mean Pi 3B) because CM3 has no built-in Bluetooth interface. CM3 isn't a variant of the Pi 3, it's a completely different board using the same SoC.

boraozgen, 2020-12-07:
You are right, but still the documentations seems to be written for all RPi based boards including CM, however one cannot follow it for CM due to the missing DT modification step. If this is supposed to be an implicit knowledge, you are right.

pelwell, 2020-12-07:
I think the Compute Module + CMIO combination is being used by more end-users than I would have expected; I have no objection to adding a user-friendly comment to the UART documentation.

michaelstoops, 2021-03-18:
I also found the UART documentation confusing, particularly the invention of "primary" and "secondary" UARTs, in addition to their hardware numbering, device types, device node paths symbolic links.

I have a draft of a major revision for that document, but I want to step away a time and proofread it before I submit the changes.

michaelstoops, 2021-03-20:
I posted a pull request: https://github.com/raspberrypi/documentation/pull/1849

commit 96142255 (2021-07-10): Add CM-specific details

Fixes #1754
closed 2021-07-10 by aallan