raspberrypi/documentation #1771: Peripheral Addresses documantation needs update for Pi 4/CM4/Pi 400

rpdom, 2021-01-03:
Please add GPIO Base address details for the Pi 4/400/CM4 range.

https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/peripheral_addresses.md

michaelstoops, 2021-03-18:
It looks like the documentation contains the result of running this function, which is the preferred way to get the values:
https://github.com/raspberrypi/userland/blob/e432bc3400401064e2d8affa5d1454aac2cf4a00/host_applications/linux/libs/bcm_host/bcm_host.c#L153

It also looks like all of the BCM2711-based devices should return the second 32-bit value from this line: https://github.com/raspberrypi/linux/blob/c9226080e513181ffb3909a905e9c23b8a6e8f62/arch/arm/boot/dts/bcm2711.dtsi#L41
```
ranges = <0x7e000000  0x0 0xfe000000  0x01800000>,
```
I have no idea why it's `0x0`.

[BCM2711 ARM peripherals datasheet](https://datasheets.raspberrypi.org/bcm2711/bcm2711-peripherals.pdf), section **5.2. Register View** says:
> The GPIO register base address is 0x7e200000.

While that should answer @rpdom 's question, I think we have some bugs. We should probably:
* Figure out whether the DT values are as they should be. I don't know enough about what "ranges" means to know whether those zeroes are supposed to be there, nor the knock-on effects of changing them.
* Adjust `bcm_host.c` to give the correct answer for BCM2711 devices.

lurch, 2021-03-18:
> Figure out whether the DT values are as they should be.

I'm fairly sure they are.

> I don't know enough about what "ranges" means to know whether those zeroes are supposed to be there, nor the knock-on effects of changing them.

I've not looked at all of the DeviceTree stuff for quite a while, but IIRC it gets a bit complicated by some fields being 32-bit, some being 64-bit, and you also have to translate child address-ranges into the address-range of their parent. Once upon a time I understood it all, and I wrote some Python code to "do it properly" https://github.com/gpiozero/gpiozero/pull/900/files#diff-c1897a36a99591e0607a33ce3fbbccac2849b4c0f9a644ce03a7f7df66056e93R112  :wink: 

> Adjust bcm_host.c to give the correct answer for BCM2711 devices.

IIRC the code in `bcm_host.c` works equally well (i.e. it gives the right answers), but it makes some assumptions and takes some shortcuts.

> The GPIO register base address is 0x7e200000.

You need to be careful to differentiate between the peripheral base address and the GPIO base address. "GPIO" is just one peripheral, out of many other peripherals.

michaelstoops, 2021-03-18:
I meant well, but I was working on this very late last night (Seattle time) and made some mistakes.

> IIRC the code in `bcm_host.c` works equally well (i.e. it gives the right answers), but it makes some assumptions and takes some shortcuts.

Yes. The catch for the 64-bit value is just two lines down: https://github.com/raspberrypi/userland/blob/e432bc3400401064e2d8affa5d1454aac2cf4a00/host_applications/linux/libs/bcm_host/bcm_host.c#L155

> You need to be careful to differentiate between the peripheral base address and the GPIO base address. "GPIO" is just one peripheral, out of many other peripherals.

You are so right.

_1.2.4. Legacy master addresses_ says:
> The Main peripherals are available from 0x7C00_0000 to 0x7FFF_FFFF

So it looks like this is indeed a just documentation issue.

PR submitted: https://github.com/raspberrypi/documentation/pull/1847

closed 2021-06-22 by aallan