raspberrypi/documentation #1772: Update the over_voltage ranges for Pi 4

worproject, 2021-01-04:
over_voltage can be set above 8 on a Pi 4 (15 seems to be the maximum)
The same limit might apply to the over_voltage_sdram_ values too.

Can someone confirm this?

lurch, 2021-01-05:
ping @popcornmix

popcornmix, 2021-01-05:
It is chip specific and subject to change if dvfs settings are tweaked. 
over_voltage adds an offset to the default voltage, which varies with chip. 
On Pi4 the minimum voltage a chip could run at is 0.83V. The max is 1.1V.
So in theory the maximum effective over_voltage is 270mV, so ~11 25mV units.
But most chips would run at a higher voltage than the minimum, so their effective range would be smaller.

As voltages are clamped to 1.1V setting a higher over_voltage will have no addional effect.

aallan, 2021-06-07:
OP should reopen if this hasn't answered your question.

closed 2021-06-07 by aallan