raspberrypi/documentation #1777: tvservice docs missing -l and -v options

6by9, 2021-01-08:
https://www.raspberrypi.org/documentation/raspbian/applications/tvservice.md
-l to list all devices and -v N to get the status of a specific device are missing.

lurch, 2021-01-08:
Fancy creating a PR? :wink:

6by9, 2021-01-08:
> Fancy creating a PR?

No time at the moment, sorry.
I had Gordon not being aware of them this morning hence creating the issue.

andrum99, 2021-01-08:
Let me see what I can do 😉

There's also a PR open on the userland repo - https://github.com/raspberrypi/userland/pull/663 could integrate - which would add man pages for `tvservice` and most of the other userland binaries.

Note that there is basic usage available from `tvservice` itself by invoking it with no parameters: this includes just about enough information to let you figure out how to use the -l and -v N options.

andrum99, 2021-01-08:
See #1778

andrum99, 2021-01-08:
Fixed by #1778

andrum99, 2021-01-08:
Fixed by https://github.com/raspberrypi/documentation/commit/94be38c47cecea2ea3551d1be24f05783c97397d

closed 2021-01-08 by JamesH65