raspberrypi/documentation #1781: Changing the default pin configuration: GPIO changes during boot

RhinoDevel, 2021-01-12:
[Changing the default pin configuration](https://github.com/raspberrypi/documentation/blob/master/configuration/pin-configuration.md) recommends (as one of three options) the following to do for peripherals, because of the different default pulls:

"Choose a GPIO pins that defaults to pulls as required by the peripheral on reset"

I assume that with "different default pulls" the default pulls set on power-on are meant.

That leads to the conclusion that no software running after these default pulls have been set (step 1) and running before applying dt-blob.bin (step 4) alter the states of the GPIOs.

If this is correct, this should be made clear on the page.

If the conclusion is wrong (between step 1 and 4 GPIO settings may get altered) the recommendation is wrong and that should get fixed.

Either way, the documentation should be more precise in relation to GPIO config changes during boot.

JamesH65, 2021-01-12:
@pelwell @timg236 Seems to me that stages 2 and 3, if they wanted to could alter the pull state of the pins, but AFAIK, they don't? Do you think clarification is needed here>

pelwell, 2021-01-12:
For what ought to be obvious reasons, the boot ROM and bootcode.bin/EEPROM by default do the absolute minimum in order to boot the device. This includes not messing with pin pulls and functions, because that would only make it harder to create a glitchless pin for use by external hardware.

It is at least worth mentioning that the EEPROM on the Pi 4, 400 and CM4 is a replacement for bootcode.bin, and that it interprets and applies any relevant `gpio=...` config settings. This will happen before `dt-blob.bin` is parsed by `start.elf`.

timg236, 2021-01-12:
The Pi4 bootloader applies gpio= and conditional filters in both the EEPROM config and when config.txt is parsed. By default, it does nothing and avoids changing GPIOs except for LED, SPI, RGMII setup

RhinoDevel, 2021-01-15:
Meanwhile I tested this with a little bare metal app and was wondering, why the GPIO pin I used for testing with an LED gets reset to its default state, even if I use soft-reset via watchdog reset or using the reset pins.

That was when I realized that I made a mistake - I interpreted "_default **pulls**_" in your documentation as "_default **GPIO settings**_"...

I **apologize** for the noise and thank you very much for the very helpful comments!

closed 2021-01-15 by RhinoDevel