raspberrypi/documentation #1787: Documentation /documentation/blob/master/installation/installing-images/mac.md contains an error

kevinnivek, 2021-01-17:
In the document, the following snippet : 

`sudo dd bs=1m if=path_of_your_image.img of=/dev/rdiskN; sync`

Should be written as : 

`sudo dd bs=1m of=path_of_your_image.img if=/dev/rdiskN; sync`

The "if" and "of" in the snippet are backwards.

lurch, 2021-01-17:
What makes you think they're backwards? "if" = "Input File" and "of" = "Output File", so for copying a disk-image onto an SD-card, `if=path_of_your_image.img of=/dev/rdiskN` looks correct to me?

michaelstoops, 2021-03-18:
To clarify, I think you're talking about:
https://github.com/raspberrypi/documentation/blob/d3e51178b5f27b90efb9f8ac62b269c8748c4b35/installation/installing-images/mac.md#L64

I agree with @lurch and the existing documentation. The context is for writing an image from file to SD card, so the `if` is the image file and `of` is the block device.

I think this issue can be closed.

lurch, 2021-03-22:
Closing as invalid.

closed 2021-03-22 by lurch