raspberrypi/documentation #1789: "Securing your Raspberry Pi" has misleading information

Ndolam, 2021-01-19:
The "Installing fail2ban" section claims  "... This needs to be copied to jail.local to enable it."
That is not true.  Just installing fail2ban gets fail2ban happily working for sshd.
Further, the section in the conf file for sshd is [sshd], not [ssh].

Thanks.

Minxers, 2021-01-26:
> The "Installing fail2ban" section claims "... This needs to be copied to jail.local to enable it."
> That is not true. Just installing fail2ban gets fail2ban happily working for sshd.
> Further, the section in the conf file for sshd is [sshd], not [ssh].
> 
> Thanks.

Can you provide a link to the page? Thanks!

Ndolam, 2021-01-26:
https://github.com/raspberrypi/documentation/blob/master/configuration/security.md
or
https://www.raspberrypi.org/documentation/configuration/security.md

In the "Installing fail2ban" section, both in the first code block and in the paragraph above it.

SirDifferential, 2021-01-29:
> The second code block in that section has [ssh] where it wants [sshd].

Correct, `sshd` is the right name. However, from what I gather, the section name doesn't matter if the `filter` is specified correctly, although I agree the proposed change is a good one.

> "This needs to be copied to jail.local to enable it."
> That is not true. Just installing fail2ban gets fail2ban happily working for sshd.

Something else about this, I'm finding some mixed comments about copying `jail.conf` entirely. The man page for `jail.conf` says:

>It is recommended that *.conf files should remain unchanged to ease upgrades.  If needed, customizations should be provided in *.local files.

and the `jail.conf` itself comments:

```
# YOU SHOULD NOT MODIFY THIS FILE.
#
# It will probably be overwritten or improved in a distribution update.
```

If the entire `jail.conf` is copied over into `jail.local` and the distro's fail2ban package is updated with a new `jail.conf`, the distro's own security improvements would be overwritten by whatever is in `jail.local` from the old version. For this reason it's better to only provide the minimal additions in `jail.local`. Please correct me if I'm wrong.

OliviaParamour, 2021-02-13:
> For this reason it's better to only provide the minimal additions in jail.local. Please correct me if I'm wrong.

This makes sense if fail2ban works right out the box without copying jail.conf over to jail.local. Is that confirmed?

aallan, 2021-08-13:
I don't think anyone reviewing the issues has the experience with fail2ban to check this. If someone can provide a pull request that'd be appreciated.