raspberrypi/documentation #1802: NOOBS [ https://www.raspberrypi.org/documentation/installation/noobs.md ]

DeeJay, 2021-01-26:
The NOOBS Installer is no longer available for download from the Software page.

The now-orphaned NOOBS installation guide https://www.raspberrypi.org/documentation/installation/noobs.md says -

"Download
Alternatively, NOOBS is available for download on the Raspberry Pi website: raspberrypi.org/downloads"

There does not seem to be a relevant link on that page, since Raspberry Pi Imager is now the preferred/promoted installation tool.

vj-codes, 2021-04-07:
@aallan @lurch @JamesH65 @DeeJay hey I would like to work on this please, suggest removing the alternative part completely as we can't download it from the website anymore, what say?

JamesH65, 2021-04-08:
I've made a couple of changes that should help push people towards Imager, but if anyone see's anywhere else that NOOBS is still highlighted, please report here.

aallan, 2021-04-08:
Going forward I don't see the NOOBS documentation making the cut when we're revising the repo to the new documentation-ng structure. We're going to be sliming down the documentation considerably, so it'll be a good opportunity to remove deprecated stuff like this.

lurch, 2021-04-08:
I think this might have been "fixed" by https://github.com/raspberrypi/documentation/commit/43e973420515e9895a228314f5918585393a4d14  :shrug:

vj-codes, 2021-04-08:
@lurch oh yeah, this issue and #1815 can be closed then

closed 2021-06-07 by aallan