raspberrypi/documentation #1815: Website: Product pages still recommend NOOBS

andrum99, 2021-02-06:
(I'm filing this here since there's no github repo for the website, and I think it's best to have this somewhere other than on the forums).

The website currently has various product pages which still recommend NOOBS, namely:

- Pi 4B - https://www.raspberrypi.org/products/raspberry-pi-4-model-b/
- Pi 3A+ - https://www.raspberrypi.org/products/raspberry-pi-3-model-a-plus/
- Pi 3B+ - https://www.raspberrypi.org/products/raspberry-pi-3-model-b-plus/
- Pi 3B - https://www.raspberrypi.org/products/raspberry-pi-3-model-b/
- Pi 1B+ - https://www.raspberrypi.org/products/raspberry-pi-1-model-b-plus/
- Pi Zero W - https://www.raspberrypi.org/products/raspberry-pi-zero-w/
- Pi Zero - https://www.raspberrypi.org/products/raspberry-pi-zero/

(I didn't miss the Pi 2B off the list - the product page for the 2B doesn't mention NOOBS)

My suggestion would be that rather than just remove reference to NOOBS from the above pages, that instead the pages be edited down to remove the section which contains the "what you will need" section which has mention of NOOBS. Alternatively, develop an alernative "what you will need" section and use that on all of the above pages (and Pi 2B, 400 and 400 kit as well).

It may also be worth tweaking the website so that each of the Pi computers uses basically the same product page, but tweaked for each specific device. This would help keep the product pages consistent, but that may not be possible with the current CMS.

lurch, 2021-02-07:
To expand on this slightly: Some of the product pages mentioned above link to https://www.raspberrypi.org/downloads/noobs/ and some link to https://www.raspberrypi.org/downloads/ but these now both redirect to https://www.raspberrypi.org/software/ which has **no** mentions of NOOBS, which _might_ confuse a new user looking for NOOBS?

andrum99, 2021-02-07:
> To expand on this slightly: Some of the product pages mentioned above link to https://www.raspberrypi.org/downloads/noobs/ and some link to https://www.raspberrypi.org/downloads/ but these now both [...]

I was more interested in there still being reference to NOOBS, but there is certainly also potential for confusion due to the issue you have highlighted.

JamesH65, 2021-02-07:
Have passed on to the Comms team, who deal with this stuff.

andrum99, 2021-06-07:
> Have passed on to the Comms team, who deal with this stuff.

Still outstanding.

aallan, 2021-06-07:
There is an effort in parallel to #1911 to update the website, and it's likely that this will fall into that bucket. I'll point the relevant folks at this one. Closing here, but I'll open in internal ticket. Thanks!

closed 2021-06-07 by aallan