raspberrypi/documentation #1818: Website: getting started guide doesn't mention Pi 400

andrum99, 2021-02-06:
The website contains an excellent 'getting started guide' which tells you all you need to know about how to get started with setting up a Raspberry Pi. Unfortunately it fails to mention the Pi 400, and includes the following text:

> There are several models of Raspberry Pi, and for most people Raspberry Pi 4 Model B is the one to choose. Raspberry Pi 4 Model B is the newest, fastest, and easiest to use.

andrum99, 2021-02-06:
Also on page 2 (at https://projects.raspberrypi.org/en/projects/raspberry-pi-setting-up/2) it has out of date screenshots of the website, and says this:

> Note: More advanced users looking to install a particular operating system should use _this guide to installing operating system images_.

which links to https://www.raspberrypi.org/documentation/installation/installing-images/README.md.

This ignores the fact that Raspberry Pi Imager will quite happily install other OSes, including an image you supply yourself. It would be better to just leave the user to use Raspberry Pi Imager, perhaps with a pointer to the bottom of the OS menu, where you can select your own OS image.

Also, you probably don't need the warning about Windows 10 / Windows Defender blocking Raspberry Pi Imager - it shouldn't do that anymore.

aallan, 2021-03-17:
See #1810.

The documentation will be undergoing an major overhaul later this year.

andrum99, 2021-03-20:
> The documentation will be undergoing an major overhaul later this year.

Agreed, however the page I reference is not part of the documentation repo - it appears to be produced by the Foundation.

aallan, 2021-05-19:
> Agreed, however the page I reference is not part of the documentation repo - it appears to be produced by the Foundation.

Ah. In which case you need to talk to the Foundation, they're a separate organisation to Trading who maintain product documentation.

closed 2021-05-19 by aallan

lurch, 2021-05-19:
There's a "Send feedback" button at the bottom of each of the Foundation-produced project pages.