raspberrypi/documentation #1824: My Touches on Official Touchscreen are Way Off

wjm103052, 2021-02-10:
I have a new raspi 4 and an official 7" touchscreen. The touches are ok in the upper left corner and the error increases as you go to the lower right where it is 1/2 inch off. This makes the screen unusable as you can't even close a window.

I have followed all of the troubleshooting tips and the problem persists. Any help to solve this would be appreciated.

Bill

wjm103052, 2021-02-10:
The problem is with my openplotter app and not with regular pi os.

closed 2021-02-10 by wjm103052

lurch, 2021-02-10:
Thanks for closing the issue so promptly, but in future questions like this are better suited for the https://www.raspberrypi.org/forums/ rather than the documentation repo :slightly_smiling_face: