raspberrypi/documentation #1834: document ipv6 routed access point

jimmont, 2021-03-04:
please expand the material in the following documentation to include details for routing ipv6, thank you
https://www.raspberrypi.org/documentation/configuration/wireless/access-point-routed.md
https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point-routed.md

JamesH65, 2021-03-04:
This will be a useful addition, however, I don't believe we have anyone in house with the relevant experience and/or time to do this at the moment.

I would be more than happy to accept a PR from anyone who does have that experience.

jimmont, 2021-03-04:
Thanks @JamesH65 I started working through the details and if successful plan to open a PR (provided no one beats me to it).

aallan, 2021-06-07:
Is this issue resolved by #1318?

lurch, 2021-07-10:
> Is this issue resolved by #1318?

Nope - #1318 seems to be about forwarding an external IPv6 connection into your internal network (e.g. if you want to make a webserver running on your Pi reachable by the whole internet), whereas this seems to be asking for details of setting up an internal IPv6 wireless network?

closed 2021-10-24 by github-actions[bot]