raspberrypi/documentation #1838: RPI 4 (not compute module) usb ports explanation

Consti10, 2021-03-08:
After reading through the "overview" of the usb ports [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/usb/README.md) I couldn't really understand what's going on.
Especially this part:

> All four USB ports on the device are connected to the USB 2.0 hub, whilst the USB 3.0 ports (blue) are ALSO connected to the USB 3.0 bus via the USB 3.0 specific pins in the socket.

But after looking at this diagram for the [VL805](https://www.via-labs.com/product_show.php?id=48) I think I know what's going on.
Please correct me if I'm wrong:

The VL805 has 4 x USB 3.0 ports
Of these 4 ports only 3 of them are SS capable (the other one is only HS/FS/LS capable)

On the rpi 4, all 4 usb ports are connected to the VL805

BUT while theoretically 3 of them could be SS, on the rpi 4, only 2 of them have been routed such that SS is available.(aka 3.0)

And the remaining 2 are only max HS (aka 2.0)

lurch, 2021-03-08:
ping @jimbojr

Consti10, 2021-03-08:
Now that I wrote it down, It looks as if I read the diagram wrong 
The diagram links all 4x usb 3. ports with both the "SS port routing" block as well as with the " HS/FS/LS block".

JamesH65, 2021-03-08:
I wrote that paragraph I think. Using lsusb --tree might help explain.  The VL805 has an internal USB2 hub, which all the USB2 plug pins are connected to, then that connects to the rest of the VL805 with a USB2 link and are limited by that.  BUT, two of the plugs also have USB 3 pins in them, which are connected to the VL805 using two of its USB3 connections, so they can run at the full PCI speed.

Consti10, 2021-03-08:
Does this mean: If you connect 2 USB 2.0 HS devices into the rpi 4, they are limited by one USB 2.0 HS link in the end.
No matter which port.
Because for this scenario, it would look like this:
Device 0 |          
               | -----> USB 2.0 (inside  VL805)  pcie -------> rpi soc
Device 1 |          

Instead of 
Device 0 |  -----> USB 2.0 (inside  VL805)  
               |                                                     pcie -------> rpi soc         
Device 1 |   -----> USB 2.0 (inside  VL805

( I hope you can see what I mean)

JamesH65, 2021-03-08:
That is correct. From the USB page.

> All connected USB 2.0 devices are connected via an internal hub which connects to the upstream PCIe link via a single USB 2.0 bus, giving a maximum combined bandwith for all USB 2.0 devices of 480Mbits/s. All four USB ports on the device are connected to the USB 2.0 hub, whilst the USB 3.0 ports (blue) are ALSO connected to the USB 3.0 bus via the USB 3.0 specific pins in the socket. USB 3.0 devices are constrained only by the total bandwidth available over the PCIe link.

Consti10, 2021-03-08:
thanks I got it. Only question remaining:

> provided by a Via Labs VL805 chip - that supports two USB 2.0 ports and two USB 3.0 ports

Should it be

 provided by a Via Labs VL805 chip - that supports 4x USB 3.0 ports of which 2 are routed as 3.0 speed on the rpi.
 
 (But you are probably leaving that as is in the documentation, else it gets complicated for the average user :) )

JamesH65, 2021-03-08:
Might be worth clarifying we don't use all the features of the VL805. I'll have a think. @lurch

lurch, 2021-03-08:
I dunno - I'm not sure if making that paragraph "more technically correct" would potentially make it even more complicated than it already is?

Here's one attempt, although I'm not sure if it's any clearer than the existing text?

For the Pi 4, a fully-featured host controller drives the downstream USB ports. Downstream USB is provided by a Via Labs VL805 chip - that supports up to four USB 2.0 ports combined with up to four USB 3.0 ports. This is connected to the BCM2711 SoC using a PCIe link, which is extremely fast. Therefore, the Pi 4 does not have the same speed constraints of previous models, which means very fast data transfer rates, especially when using the USB 3.0 ports. All connected USB 2.0 devices are connected via an internal USB 2.0 hub which also connects to the upstream PCIe link via a single USB 2.0 bus, giving a maximum combined bandwith for all USB 2.0 devices of 480Mbits/s. All four USB ports on the Pi 4 are connected to this USB 2.0 hub, whilst the two USB 3.0 ports (blue) are ALSO connected to the USB 3.0 bus via the USB 3.0 specific pins in those sockets. USB 3.0 devices are constrained only by the total bandwidth available over the PCIe link.

andrum99, 2021-06-14:
I've updated #1887 to mention that the USB 2.0 lines on the Pi 4 are connected to a single USB 2.0 hub, and that this means the max bandwidth available to USB 2.0 and 1.1 devices is the same as that of a single USB 2.0 port.

closed 2021-06-17 by aallan