raspberrypi/documentation #1846: bootmodes/msd.md issue

jacksnodgrass, 2021-03-17:
In: 
documentation/hardware/raspberrypi/bootmodes/msd.md

and on: 
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

it says: 
Check if your Pi 4B has the required bootloader EEPROM version
To check if your Pi 4B has the required bootloader EEPROM version, power it up with no SD card inserted and a display attached to one of the HDMI ports. The Pi 4B will display a diagnostic screen on the attached display, which includes the bootloader EEPROM version at the top of the screen. The bootloader must be dated Sep 3 2020 or later to support USB mass storage boot. If the diagnostic screen reports a date earlier than Sep 3 2020, or there is no diagnostic screen shown, you will need to update the bootloader EEPROM first to enable USB mass storage boot.

USB mass storage boot on the Pi 4B requires Raspberry Pi OS 2020-08-20 or later.

so it would seem to say that if you have this Sep 3 2020 boot loader and a 2021 version of the Raspberry Pi OS, you should be able to boot from the USB mass storage. 

later in these instructions it says: 
After preparing the storage device, connect the drive to the Raspberry Pi and power up the Pi, being aware of the extra USB power requirements of the external drive. After five to ten seconds, the Raspberry Pi should begin booting and show the rainbow splash screen on an attached display. Make sure that you do not have an SD card inserted in the Pi, since if you do, it will boot from that first.

See the bootmodes documentation for the boot sequence and alternative boot modes (network, USB device, GPIO or SD boot).

... but when I have the correct software on my Raspberry Pi 4 8gb unit and boot from the USB device with no SD card mounted it just says install an SD card. 

[image]

I had to go into the raspberry-config stuff and set the bootloader order to boot from the USB mounted device before I could get the Pi to boot from the USB mounted device. 

The instructions seem to say that the boot order is optional... but the USB drive was not used at all until I changed the boot order. 

I am not sure why I had to mess with the boot order... no SD card was installed... a USB Drive was... and it did not try and use the USB drive untill I booted from the SD Card, changed the boot order and then booted with no SD card attached.

michaelstoops, 2021-03-18:
Hi @jacksnodgrass.

I'm sorry you had trouble with booting to your USB drive.

If I read this correctly, I think you're concerned that having the Sep 3 2020 bootloader is not enough to boot from USB. You also have to have the correct boot order set.

Does that accurately summarize your concern?

Michael

jacksnodgrass, 2021-03-18:
Yes... the docs seems to say that IF you have the Sep 3 2020 boot loader, then your all set.... it will boot from the SD card OR it will boot from the USB drive... since the default boot order is SD card, then USB drive. 

But the default boot order is more like SD Card or it won't boot. You still have to boot with an SD card to enable booting from the USB Drive. I was expecting just JUST be able to boot the first time from the USB Drive since the bootloader was the correct version.

lurch, 2021-03-18:
ping @timg236

timg236, 2021-03-18:
Sep 3 2020 is necessary but not sufficient for USB boot. You can use the Raspberry Pi Imager to set the boot-order

jacksnodgrass, 2021-03-18:
Not to be dense... but I want to make sure I understand what the issue is. 

If you take a brand new, out of the box pi 4 8gb.... can it boot from a correctly formatted USB drive if that's attached to the system without changing the boot order on the boards boot eprom?

timg236, 2021-03-18:
Pi400 is the first product to officially support USB out of the box. Pi4 including 8GB potentially needs a bootloader update (e.g. via Raspberry Pi Imager) depending upon when it was manufactured.

lurch, 2021-03-18:
I _think_ there's also the subtle issue (I'm sure @timg236 will correct me if I'm wrong!) that using Raspberry Pi Imager to update the bootloader on a Pi4 will always overwrite the eeprom-config (where the boot-order is stored) with whatever eeprom-config is stored in the bootloader; but if using the rpi-eeprom-update command to update the bootloader, then depending on what options you pass to the `rpi-eeprom-update` command it might try 'migrating' your existing eeprom-config?

lurch, 2021-07-10:
> Pi400 is the first product to officially support USB out of the box.

I think the Pi 3B+ supported USB boot out of the box too, so perhaps you meant "Pi400 is the first BCM2711-based product to officially support USB boot out of the box." ?

aallan, 2021-08-09:
Documentation source now lives at computers/raspberry-pi/boot-msd.adoc

aallan, 2021-08-10:
Unless someone wants to open a PR against the documentation I'll probably close this one shortly..?

timg236, 2021-08-10:
I think this can be closed

commit 318bafae (2021-08-10): Fixes #1846 ?
commit 31af75e5 (2021-08-10): Merge pull request #2016 from raspberrypi/lurch-patch-1

Fixes #1846 ?
closed 2021-08-10 by aallan