raspberrypi/documentation #1857: USB_MSD_PWR_OFF_TIME boot mode might be incorrectly stated

SirDifferential, 2021-03-24:
In the Raspberry Pi 4 [Raspberry Pi 4 bootloader configuration](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md), it is said:

>USB_MSD_PWR_OFF_TIME
>
>During USB mass storage boot, power to the USB ports is switched off for a short time to ensure the correct operation of USB mass storage devices. Most devices work correctly using the default setting: change this only if you have problems booting from a particular device. Setting USB_MSD_PWR_OFF_TIME=0 will prevent power to the USB ports being switched off during USB mass storage boot.

This makes it sound like:

1) This option only applies when the Raspberry boots from a USB mass storage device and not SD card. I assume this refers to boot mode 0x4, USB-MSD, and not 0x1, SD CARD.
2) The documentation also says:

>During USB mass storage boot, power to the USB ports is switched off for a short time to ensure the correct operation of USB mass storage devices. 
 
The mention of mass storage devices is a bit confusing, and again made me wonder if this setting is somehow only relevant for mass storage boot and not other USB devices. I set the setting to 5000 and put a little LED indicator device in the USB port. It looked like the USB ports were unpowered for 5 seconds when booting from an SD card. Is the documentation off here, and is this feature supposed to work regardless of what boot mode is used?

This came up when I was looking for ways to cut power to problematic USB cameras to power cycle them on boot.

lurch, 2021-03-24:
ping @timg236

SirDifferential, 2021-08-30:
Not sure if the bot action suggests this issue will be closed for inactivity. I'd still like to have a clarification about this boot mode parameter.

lurch, 2021-08-31:
ping @timg236 @peterharperuk

peterharperuk, 2021-08-31:
Yes, it does look like USB is always switched off for 1s by default regardless of the boot mode. It looks like this might be done to avoid problems with some devices that might be plugged into the device.

closed 2021-11-07 by github-actions[bot]