raspberrypi/documentation #1862: Documentation on boot/wpa_supplicant.conf may not be accurate

trbarrettjr, 2021-03-28:
Links are:
https://github.com/raspberrypi/documentation/blob/master/configuration/boot_folder.md
https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/headless.md

When I follow these two documentations to connect to a wireless network from a headless system, the Pi is not connecting to my wifi network.  This has been verified using my wifi interface and not seeing `raspberrypi` getting an address from the DHCP.

I have done a little forensics.  I see what is in `boot/wpa_supplicant.conf` is in `/etc/wpa_supplicant/wpa_supplicant.conf`; however, there is no `update_config=1` or `ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev` that we normally see.

Sorry if this is a Linux issue; I don't know where to start or report it.

I tried this procedure on the following hardware: Pi 3 and Pi 4 (4 GB of RAM).  I utilized the latest lite image (2021-03-04) from the website.

Thank you!

trbarrettjr, 2021-03-28:
After doing some experimenting with the Pi; the following lines are required to be in the `boot/wpa_supplicant.conf`

`ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev` and
`update_config=1`
then followed by:
```
network={
    ssid="private network"
    psk="my secret key"
}
```

Please close if you want, I would recommend that the https://github.com/raspberrypi/documentation/blob/master/configuration/boot_folder.md be updated to include the full code (to include the `ctrl_interface` and `update_config`).

Thank you!

lurch, 2021-03-28:
I'm confused... https://github.com/raspberrypi/documentation/blob/master/configuration/boot_folder.md just links to https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/headless.md and the latter page _does_ include the `ctrl_interface` and `update_config` lines?? Am I missing something? Is there some other reference to a `wpa_supplicant.conf` that I'm not seeing?

closed 2021-06-17 by aallan