raspberrypi/documentation #1870: bcm2711 arm local register LOCAL_PRESCALER not documented

alepi, 2021-04-13:
In https://github.com/raspberrypi/tools/blob/master/armstubs/armstub8.S I have seen register LOCAL_PRESCALER at offset 0x08 and got an idea about how it works.

It should be listed in the bcm2711 manual in section "6.5.2. ARM_LOCAL".

lurch, 2021-04-13:
I assume by "bcm2711 manual" you mean https://datasheets.raspberrypi.org/bcm2711/bcm2711-peripherals.pdf ?

@pelwell Anything useful about `LOCAL_PRESCALAR` that it's worth adding to that datasheet? Or is it something that mere mortals shouldn't mess with?

pelwell, 2021-04-13:
The ARM_LOCAL section of the BCM2711 peripherals PDF appears to be incomplete/selective. I have no knowledge as to how the selection was arrived at, and I'm not the person to decide whether it should be extended.

alepi, 2021-04-13:
I have downloaded QA7 rev 3.4.
The register is documented there at page 10.

lurch, 2021-04-13:
Although there's obviously no guarantees that the register behaves the same on BCM2836 and BCM2711, so be careful :slightly_smiling_face:

alepi, 2021-04-14:
@lurch thanks for your concern :-)

I appreciate that you wrote the documentation by following someone else's guidelines and with the need to protect someone else's intellectual property. I know how these things work.

On the other hand, please understand that I created this issue with a good reason and I thought that by posting here I was able to talk to someone who could make the call of whether to satisfy my request or not.

I am learning about aarch64 and the lack of that detail made my learning experience unnecessarily frustrating, so I thought of asking that it gets documented and thought that this was the right place to ask.

Is there any way that the person who makes this kind of decisions gets involved in this discussion ?

JamesH65, 2021-04-14:
That's a director level discussion, I'll flag it up to someone at that level.

closed 2021-10-17 by github-actions[bot]