raspberrypi/documentation #1872: The self created /boot/ssh and /boot/wpa_supplicant.conf will be deleted after reboot on Raspberry Pi 4 Model B Rev 1.4.

hongyi-zhao, 2021-04-14:
I've a Raspberry Pi 4 Model B Rev 1.4, which has the following hardware and software info:

```
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: d03114
Serial		: 10000000dd3ebf3c
Model		: Raspberry Pi 4 Model B Rev 1.4
pi@raspberrypi:~ $ cat /etc/debian_version
10.7
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

According to the notes [here](https://www.raspberrypi.org/documentation/configuration/boot_folder.md) and [here](https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md), I created an empty file `/boot/ssh` and another file `/boot/wpa_supplicant.conf` with the following content:

```
# /boot/wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=CN

network={
    ssid="qcx205"
    psk="12345678"
    priority=1
    #id_str="tl-wdr5620"
}

network={
    ssid="HUAWEI_B311_EF3A"
    psk="12345678"
    priority=2
    #id_str="b311b-853"
}
```

But I noticed that after restarting the Raspberry Pi, the created files mentioned above will be deleted automatically. Any hints for this problem will be highly appreciated.

Regards,
HY

lurch, 2021-04-14:
I think that's to be expected? `/boot/ssh` gets deleted when Raspberry Pi OS enables SSH, and `/boot/wpa_supplicant.conf` gets deleted when Raspberry Pi OS moves the file to `/etc/wpa_supplicant/wpa_supplicant.conf`. If Raspberry Pi OS _didn't_ delete the file from `/boot`, and both `/boot/wpa_supplicant.conf` and `/etc/wpa_supplicant/wpa_supplicant.conf` were edited, it wouldn't know which one to use! :wink: 
(ping also @XECDesign )

aallan, 2021-04-14:
> I think that's to be expected? 

Yes. This is the expected behaviour.

lurch, 2021-04-14:
Closing issue. Feel free to add further comments if this is causing problems.

closed 2021-04-14 by lurch