raspberrypi/documentation #1881: Raspberry Pi Compute Module 4 STEP file is a joke

EmileBV, 2021-04-20:
I'm not entirely sure if this is the right place to post this issue, but this web documentation is the only place where I could find a download link to the Compute Module 4 STEP files, so I'll assume this is correct.

This issue is concerning the STEP files for the Compute Module 4 found at the bottom of the [Mechanical Drawings page](https://www.raspberrypi.org/documentation/hardware/raspberrypi/mechanical/README.md).

The file, when imported in a CAD program such as Solid Edge, is translated to a usable format and is uncompressed, transformed into the individual components that were included in the STEP file export. This is normal behavior and completely expected, as it's what the STEP format is made for.

Where the issue comes in is that every single parts of the assembly is named `(Unsaved)` followed by a number. This could be fine on its own but the issue is every part that is repeated and **not unique** is actually individual **unique** parts. This ends up with a total file size of over 300MB and 2263 unique parts that are each individual files on the disk.
[image]

This makes the 3D model completely unusable in any scenario as this single assembly weights more than most projects it would get used in. The STEP files need to be re-created entirely and optimized to not be in such a state. A different solution could be to have a mockup file that is much simpler and only includes the critical dimensions of the model, as the current assembly contains way more detail than anyone would ever need.

There is no need for every single solder ball on every single BGA chip to be an individual unique component.
[image]

JamesH65, 2021-04-20:
@dp11 Is this an issue at our end, or something gone awry in some conversion somewhere?

lurch, 2021-04-20:
I suspect James meant to ping @dp111 ?

EmileBV, 2021-04-20:
I apologize for the harsh issue title at first, I admit I was a bit shocked when finding this out and writing the issue title.

dp111, 2021-04-20:
If you are unhappy at how file imports to your CAD system, It might be worth contacting you CAD supplier.  The STEP file is a detailed model and gets used for other purpose where the extra details is useful. If you want less detail then you can delete items as you have done.

EmileBV, 2021-04-20:
It's not a question of details, it's the individual files that were exported into the STEP format have no descriptive name and way too many components are left as unique components when they should have been grouped into a single part from the start. In the many years I've been using STEP files from various sources and of varying complexity, I've never seen something like this where individual solder beads were individual unique parts rather than instances or a larger group containing all the instances.

This model could be as detailed as it currently is without having so many parts if some grouping was used or parts got re-used multiple times.

If this is impossible to achieve, could a mock-up of the critical dimensions be made available instead?

dp111, 2021-04-20:
You are welcome to create a mock-up version by deleting items that are of no interest to you. Please also take into account the note in the datasheet page 15 about the step files.

aallan, 2021-06-08:
Closing. Unlikely to be further progress.

closed 2021-06-08 by aallan