raspberrypi/documentation #1884: Documentation for Raspberry Pi OS is still filed under raspbian

mrlowndes, 2021-04-21:
Could we change the path to Documentation > Raspberry Pi OS instead?

[image]

JamesH65, 2021-04-21:
@mudge This is one for you I think.

lurch, 2021-04-21:
@aallan is going to be having a big docs refactoring at some point, so this might be better tackled as part of that?

JamesH65, 2021-04-22:
Now that I think about it, this rings a bell. I think to prevent external links to the docs breaking, we left that folder named Raspbian.

aallan, 2021-04-22:
Yup. This is already on my list to fix during the `documentation-ng` refactor.

closed 2021-04-22 by aallan

aallan, 2021-04-22:
> Now that I think about it, this rings a bell. I think to prevent external links to the docs breaking, we left that folder named Raspbian.

Yes. We can get @mudge to fix up the redirects when the time comes to deploy.