raspberrypi/documentation #1885: wpa_supplicant.conf and ssh files in boot folder no longer working for Raspberry Pi OS?

mrlowndes, 2021-04-21:
Tried putting these two files on a freshly imaged SD card in the boot directory for a headless set up and it did not work. Had to go get the Pi and connect it to a monitor, mouse, and keyboard - very annoying

JamesH65, 2021-04-21:
They should both still be working. We've not had any other reports as yet to indicate a general problem.

mrlowndes, 2021-04-21:
I don't know what I could have done differently? When I plugged it into my monitor, it hadn't enabled ssh or connected to my network. Had to go through the initial setup on the desktop...

lurch, 2021-04-21:
Do you still see those files in `/boot` on your SD card?

mrlowndes, 2021-04-21:
I don't know, I had to install it for my application asap

lurch, 2021-04-22:
If you don't have time to help investigate the issue (which is totally understandable), then there's probably much we can do to help? :man_shrugging:

golovasteek, 2021-05-08:
I have a similar problem: rapspberry pi zero doesn't connect to the wifi after boot.
Steps I've done:
  * flash raspbian lite
  * added wpa_supplicant.conf into /boot folder on the SD card (namely /Volumes/boot/boot, since I use macOS)
  * started raspberry pi zero with this card
  * wait a little ( 5 minutes)
Raspberry pi still doesn't connect to WIFI

after checking the SD card content again, the wpa_supplicant.conf still in the same place.

Hope this can help.

golovasteek, 2021-05-08:
After playing around it turns out the the `boot` folder on the SD card is not needed. And wpa_supplicant.conf should be placed at the root of SD card.

lurch, 2021-05-08:
You appear to have misunderstood the instructions - I believe in your case you'd want to place the file into `/Volumes/boot` rather than `/Volumes/boot/boot` ?
There are actually two partitions on the SD card - boot (FAT32) and root (ext4), and I believe Windows and Mac only show the contents of the FAT32 partition, as they're not able to read ext4 filesystems.

golovasteek, 2021-05-08:
Now, when you mentioned it, I see the second partition I see it.

There are couple of things, which can be improved.
  * the WiFi documentation mentions "[boot folder](https://www.raspberrypi.org/documentation/configuration/wireless/headless.md)", which is different than "[boot partition](https://www.raspberrypi.org/documentation/remote-access/ssh/)" (whichis correctly mentioned in the SSH instruction"
  * the fact that there are two portions, not easy to find at least for newbees. I for example googled the "raspberry pi headless" and found the [page](https://www.raspberrypi.org/documentation/configuration/wireless/headless.md). 
  
  Now, when I read it carefully, I see that it references the "[boot folder](https://www.raspberrypi.org/documentation/configuration/boot_folder.md)" document, which kind of explains it.
  
 But before the system is booted on raspberry pi, `boot` is not folder, but partition.
 
 So I propose to change wording in the "headless setup" to "root of boot partion"

lurch, 2021-05-10:
Fixed (hopefully?) by #1899

closed 2021-05-10 by lurch