raspberrypi/documentation #1888: Erratic revision codes

cr-chsn1, 2021-04-23:
Since the revision codes are unique I try to identify the used RPi-model by looking at this code.
Unfortunatelly the codes I read out of my different models do and don't match with the ones provides in the official documentation.

Link: https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md

The OS used for testing:
> Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
> Feb 25 2021 12:12:09
> Copyright (c) 2012 Broadcom
> version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)

These models were tested:
| 						Model					| Rev Code (documentation)	| Rev Code (read out) 	|
| --------------------------------------------- | ------------------------- | --------------------- |
| Raspberry Pi Model B Rev 2 (512 MB)			| 000e						| ***100000e***			| 
| Raspberry Pi Model B+ Rev 1.2 (512 MB)		| 900032 					| ***1900032***			|
| Raspberry Pi 2 Model B Rev 1.1 (1 GB) 		| a01041 					| ***2a01041*** 		|
| Raspberry Pi 3 Model B Rev 1.2 (1 GB) 		| a02082 					| ***2a02082*** 		|
| Raspberry Pi 3 Model B+ (1 GB) 				| a020d3 					| a020d3 				|
| Raspberry Pi 4 Model B Rev 1.1 (1 GB) 		| a03111 					| a03111 				|
| Raspberry Pi Compute Module 4 Rev 1.0 (1 GB)	| a03140 					| a03140 				|

On my later models, the revision code is like the one shown in the official documentation (new-style code, 6 digits).
My oldest RPi has no old-style code and it's a 7 digit one. On RPi1 a ***1*** is prefix to a new-stylish code and on my RPi2/3 a ***2*** is prefix to the revision code.

Is the documentation outdated (no old-style code today) and what's that 7 digit code on the older RPis?

Some screenshots:
[image]
[image]
[image]

lurch, 2021-04-24:
> 100000e

The leading 1 indicates that "Warranty has been voided by overclocking".

> 1900032

The [revision-codes](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md) table says the leading one corresponds to one of the "unused" bits. Does this bit have a use now @pelwell ?

> 2a01041 / 2a02082

The leading 2 corresponds to the "`W` Warranty bit" in the table where the `NOQuuuWuFMMMCCCCPPPPTTTTTTTTRRRR` bit-pattern is decoded.

cr-chsn1, 2021-04-24:
Since almost every RPi at home should have a voided warrenty, why do the 4B and CM4 behave like there is no warrenty bit?

**Example: Raspberry Pi CM4**

`vcgencmd get_config force_turbo && vcgencmd get_config arm_freq && vcgencmd measure_clock arm && vcgencmd get_config over_voltage && vcgencmd measure_volts core`
```
force_tubro=1
arm_freq=2300
frequency(48)=2300537088
over_voltage=12
volt=1.1000V
```

This should be enough to void the warrenty.

**Result:** 
`cat /proc/cpuinfo | grep 'Revision'`
```
Revision        : a03140
```

I also had this idea, but since my 4B and CM4 seem to ingnore it, I withdraw this thought.

My RPi1B has the revision code **100000e** which is 1000000000000000000001110.
 
| W | F | MMM | CCCC | PPPP | TTTTTTTT | RRRR |
| --- | --- | --- | --- | --- | --- | --- |
| 1 | 0 | 000 | 0000 | 0000 | 00000000 | 1110 |

It's a Model B (not A) with 512 MB (not 256 MB) of RAM and the *E* for revision doesn't make sense to me either.

Maybe I have don't get the behaviour right. Do I have a false conception?

lurch, 2021-04-24:
The RPi1B uses the "Old-style revision codes" not the "New-style revision codes".

cr-chsn1, 2021-04-24:
If bit F = 0, then only the last 4 bits are important and everything else is irrelevant and filled with 0s. Okay, I understand.
But the preceding ***1*** from 100000e and 1900032 seems to be warrenty bit set to 1, because of the overclocking. Am I right?

Then are two questions unanswered:
1. What about the 2 preceding my RPi2B and 3B?
2. Why is there no warrenty bit set to 1, when I overclock a RPi3B+, 4B, CM4?

lurch, 2021-04-24:
> But the preceding 1 from 100000e seems to be warrenty bit set to 1, because of the overclocking. Am I right?

Yes, as I already said: The leading 1 indicates that "Warranty has been voided by overclocking".

> 1900032

This is unusual, because it's a new-style revision code, but the 1 corresponds to an "unused" bit (between the `W` and `F` bits).

> What about the 2 preceding my RPi2B and 3B?

I already answered that above: The leading 2 corresponds to the "`W` Warranty bit" in the table where the NOQuuuWuFMMMCCCCPPPPTTTTTTTTRRRR bit-pattern is decoded.

> Why is there no warrenty bit set to 1, when I overclock a RPi3B+, 4B, CM4?

I dunno the answer to that, but perhaps @popcornmix does?

cr-chsn1, 2021-04-24:
Ah OK. Thanks for clarification.

The 1900032 comes from the new-style RPi1B+. This one behaves like documented.
This model seems to have an old and a new style revision code: 0010 vs. 900032

1900032 = 1100100000000000000110010

| W | F | MMM | CCCC | PPPP | TTTTTTTT | RRRR |
| --- | --- | --- | --- | --- | --- | --- |
| 1 | 1 | 001 | 0000 | 0000 | 00000011 | 0010 |
| OCed | New-Style | 512 MB | Sony UK | BCM2835 | B+ | Rev. 1.2 |

Maybe the others know why unused bit are being used. And why there is this odd behaviour of the unvoidable RPis.

lurch, 2021-04-24:
> This model seems to have an old and a new style revision code: 0010 vs. 900032

Nah I think it's just coincidence that the new-style revision of binary `0010` happens to look the same as the old-style model of hexadecimal `0010` :wink:  (unless I've misunderstood what you're saying?)
But yes, "old" RPi1B+ will have an old-style revision code of 0010 and "new" RPi1B+ will have a new-style revision code of 900032 - just depends on when they were manufactured.

cr-chsn1, 2021-04-25:
No, I tried to say, that this Raspberry Pi has an old-style revision code: 0010
[image]

...and a new-style revision code: 900032
[image]

Otherwise the F-bit wouldn't be necessary IMO.

popcornmix, 2021-04-26:
We don't set warranty bit on Pi4/CM4 (we've seen no correlation with warranty bit being set and increased chip failures on older models of Pi).

lurch, 2021-04-26:
Does this answer all your questions @cr-chsn1 ? Can this be closed now?

lurch, 2021-04-26:
I just added a small tweak: https://github.com/raspberrypi/documentation/commit/b021613bc0c59b5a31557480092de987031bb110

cr-chsn1, 2021-04-26:
Everything beside the 2, which comes from the unused bit is answered.
Since I read the last 6 digits from right to left, it won't affect my identification if there is a 1 or a 2 before that.

Maybe this will be useful to somebody in the future:
`revision=$(tr -d '\0' < /proc/cpuinfo | grep "Revision" | rev | cut -c 1-6 | rev)`

Thanks for clarification and your patience!

closed 2021-04-26 by cr-chsn1

lurch, 2021-04-26:
> Everything beside the 2, which comes from the unused bit is answered.

It's the leading `1` that comes from the unused bit, not the leading `2`. Decoding the NOQuuuWuFMMMCCCCPPPPTTTTTTTTRRRR bit-pattern:

| Hex-string | N | O | Q | uuu | W | u | F | MMM | CCCC | PPPP | TTTTTTTT | RRRR |
| --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- |
| 1900032 | 0 | 0 | 0 | 000 | 0 | 1 | 1 | 001 | 0000 | 0000 | 00000011 | 0010 |
| 2a01041 | 0 | 0 | 0 | 000 | 1 | 0 | 1 | 010 | 0000 | 0001 | 00000100 | 0001 |

> Since I read the last 6 digits from right to left

Note that the table of "New-style revision codes in use" at https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md isn't necessarily complete - there may be other revision-codes in use which _don't_ appear in that list (e.g. all the different memory-configurations of CM4).
So to make your code "robust" you need to pick apart the individual fields from within the bit-string, rather than trying to simply match on the last 6 hex-characters. https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update#L300 is probably a good example of how you can do that in bash.

cr-chsn1, 2021-04-26:
Now I understand :)

An associative array filled with data does the trick very well. It's nothing mission critical.