raspberrypi/documentation #1889: Not one document on www.raspberrypi.org seems to point out camera connector for Pi 4B

FolsomMike, 2021-04-24:
Every picture which points out different components ignores the camera port and the display port. Only the schematic gives any info (as in "J3") which is not useful if the computer is in a case and cannot be examined for reference designators.

Main Web Page ~ NOPE
https://www.raspberrypi.org/products/raspberry-pi-4-model-b/

Tech Specs ~ NOPE
https://www.raspberrypi.org/products/raspberry-pi-4-model-b/specifications/

Product Brief  ~ NOPE
https://datasheets.raspberrypi.org/rpi4/raspberry-pi-4-product-brief.pdf

Schematics TELLS YOU IT'S J3 (which is minimally useful)
https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_4b_4p0_reduced.pdf

Mechanical ~ NOPE
https://www.raspberrypi.org/documentation/hardware/raspberrypi/mechanical/rpi_MECH_4b_4p0.pdf

lurch, 2021-04-24:
Does this help? https://www.raspberrypi.org/documentation/usage/camera/installing.md

FolsomMike, 2021-04-24:
Even that page can't seem to point to the connector! You have to watch a video. I hope they don't replace all documentation with videos!

lurch, 2021-04-24:
Haha :slightly_smiling_face:  I think in this case a video is actually appropriate though, as if somebody's never used an FFC connector before they might not know how to use it? (they can be quite delicate)
Assigning to @aallan for the final decision.

aallan, 2021-04-24:
I'll leave a note to myself about this for the upcoming documentation review.

closed 2021-04-24 by aallan