raspberrypi/documentation #1892: *** Can't find default configuration "arch/arm64/configs/bcm2709_defconfig"!

Marietto2008, 2021-04-29:
root@zi-desktop:~/Desktop/linux# KERNEL=kernel7
root@zi-desktop:~/Desktop/linux# make bcm2709_defconfig
***
*** Can't find default configuration "arch/arm64/configs/bcm2709_defconfig"!

JamesH65, 2021-04-29:
Not a documentation issue, please try the forums. You will also need to provide more details on what you are doing to get this error.

closed 2021-04-29 by JamesH65

lurch, 2021-04-29:
At a quick guess it looks like you're trying to compile a 32-bit Pi kernel whilst running on a 64-bit Pi OS. You need to follow the cross-compiling instructions to do that.

Marietto2008, 2021-04-29:
I'm trying to compile a 32 bit kernel on my jetson nano. but I have added the armhf architecture with this command ; # dpkg --add-architecture armhf. The instructions that I'm reading are here : https://www.raspberrypi.org/documentation/linux/kernel/building. It does not explain well what are the circumstances within which this command !"make bcm2709_defconfig" is valid. You can argue that I should know what I'm doing. But it's wrong because you can't expect that everyone has a master degree. I'm trying to do my best,but my job is the psychologist.

lurch, 2021-04-29:
> The instructions that I'm reading are here : https://www.raspberrypi.org/documentation/linux/kernel/building.md
I have added the armhf architecture with this command ; # dpkg --add-architecture armhf.

Those instructions say nothing about `--add-architecture`??

The instructions for "Local building" clearly say "On a Raspberry Pi...". For everything else, you'll need to follow the "Cross-compiling" instructions.

Marietto2008, 2021-04-29:
I don't have a raspberry pi,but I know that I can run the armhf applications at 32 bit on the jetson nano at 64 bit.

aallan, 2021-04-29:
This repo is the documentation repo for the Raspberry Pi…

> I'm trying to compile an armhf 32 bit kernel on my jetson nano.

…we really can't help you. I'd probably ask on the NVIDIA forums.

lurch, 2021-04-29:
If you don't have a Raspberry Pi, then I've got no idea why you're trying to build a bcm2709 kernel? (which, as the page says, is specific for "Raspberry Pi 2, Pi 3, Pi 3+, and Compute Module 3") :confused: 

I've got no idea what you're trying to do, but you're probably better off asking for help on the jetson nano forums?

Marietto2008, 2021-04-29:
because I'm trying to run x86 on arm32 bit following this tutorial :

https://www.novaspirit.com/2019/04/15/run-x86-arm/

this tutorial is called by another tutorial,this : https://forums.developer.nvidia.com/t/installing-wine-on-jetson-nano/170095

aallan, 2021-04-29:
Here is not the place to ask for help. Try the NVIDIA forums.