raspberrypi/documentation #1893: DSI Touchscreen pages still reference ft5406

6by9, 2021-05-03:
https://github.com/raspberrypi/documentation/blob/master/hardware/display/troubleshooting.md references grepping for ft5406 to confirm the touchscreen is working.

The module was renamed to raspberrypi-ts when it was upstreamed in 2018 as part of the 5.0 release.

andrum99, 2021-05-04:
On latest Raspberry Pi OS with kernel 5.4.35-v7l+ there's no such module. I do however have one named `/lib/modules/5.4.35-v7l+/kernel/drivers/input/touchscreen/edt-ft5x06.ko`.

I've decompiled the rpi-ts5406.dtbo file and it provides no useful information as to which kernel module the screen uses. Perhaps someone in the known could confirm that the above is the correct module: I will then file a PR to update the docs.

6by9, 2021-05-04:
5.4.35 is not the latest kernel.

https://github.com/raspberrypi/firmware/tree/master/modules/5.10.33-v7l%2B/kernel/drivers/input/touchscreen lists 
raspberrypi-ts.ko

andrum99, 2021-05-04:
> 5.4.35 is not the latest kernel.
> 
> https://github.com/raspberrypi/firmware/tree/master/modules/5.10.33-v7l%2B/kernel/drivers/input/touchscreen lists
> raspberrypi-ts.ko

Apologies - I was trying to verify and chose the wrong version from the directory listing. I've corrected the PR per your original statement.

aallan, 2021-06-22:
Closed associated PR, so closing the issue.

closed 2021-06-22 by aallan