raspberrypi/documentation #1898: [help]UFW, iptables, firewalld can not enable port, do not know how to enable

Conmi-WhiteJoker, 2021-05-06:
I tried using them to open port 3306, but never succeeded.However, if SSH is checked in the "raspi-config", port 22 will open normally.
I don't know if the Raspberry Pi has updated any firewall Settings, it used to be able to open port 3306 directly using UFW.
I might need a little help.
Thank you.

Conmi-WhiteJoker, 2021-05-06:
Can someone give me a hand?

aallan, 2021-05-06:
This is the sort of question you should be asking on the [forums](https://www.raspberrypi.org/forums/) as this repo is for documentation, so I’m closing this issue. But please follow up with further comments if you have specific questions about the [documentation](https://www.raspberrypi.org/documentation/).

closed 2021-05-06 by aallan

Conmi-WhiteJoker, 2021-05-06:
> 这是你应该在[论坛上](https://www.raspberrypi.org/forums/)问的问题， 因为这个回购是文档， 所以我要结束这个问题。但是，如果您对[文档](https://www.raspberrypi.org/documentation/)有具体问题，请跟进进一步的评论。

Okay, thank you.

Conmi-WhiteJoker, 2021-05-06:
> This is the sort of question you should be asking on the [forums](https://www.raspberrypi.org/forums/) as this repo is for documentation, so I’m closing this issue. But please follow up with further [...]

Okay, thank you.