raspberrypi/documentation #1900: `libcamera` install failed when following the document

H-xie, 2021-05-13:
The document I  #followed: [libcamera and libcamera-apps installation for Raspberry Pi ](https://github.com/raspberrypi/documentation/blob/master/linux/software/libcamera/README.md )

The raspberry pi version: **Zero W**
OS version: **Raspberry Pi OS Lite** (release date: March 4th 2021)

The solution I found lists as follow:

1. `pyyaml ` needs specifiy version `sudo pip3 install pyyaml==5.4`. Since piwheel.org failed to compile the latest ver. 5.4.1, the installation will failed.
2. `jinja2` needs specify version `sudo pip3 install jinja2==2.11.3`. Since the latest version of jinja2 (3.0.0) change the API, `meson -C build` will failed with error
```shell
ninja: Entering directory `build'
[1/114] Generating mojom_templates with a custom command
FAILED: utils/ipc/libcamera_templates.zip 
/home/pi/libcamera/utils/ipc/./generate.py -o utils/ipc precompile
Traceback (most recent call last):
  File "/home/pi/libcamera/utils/ipc/./generate.py", line 29, in <module>
    generator.main()
  File "/home/pi/libcamera/utils/ipc/mojo/public/tools/bindings/mojom_bindings_generator.py", line 385, in main
    return args.func(args, remaining_args)
  File "/home/pi/libcamera/utils/ipc/mojo/public/tools/bindings/mojom_bindings_generator.py", line 258, in _Precompile
    template_expander.PrecompileTemplates(generator_modules, args.output_dir)
  File "/home/pi/libcamera/utils/ipc/mojo/public/tools/mojom/mojom/generate/template_expander.py", line 83, in PrecompileTemplates
    ignore_errors=False)
TypeError: compile_templates() got an unexpected keyword argument 'py_compile'
[2/114] Generating control_ids_h with a custom command
ninja: build stopped: subcommand failed.
```

---

I suggest add version specification of all dependencies in the document to it easy to follow. 
Thank you for your time.

JamesH65, 2021-05-13:
@davidplowman @naushir Any thoughts?

JamesH65, 2021-05-13:
@bennuttall For the piwheels reference.

bennuttall, 2021-05-13:
Fixed missing PyYAML: https://piwheels.org/project/PyYAML/ (@H-xie please report these to piwheels in future)

JamesH65, 2021-05-13:
Thanks @bennuttall

davidplowman, 2021-05-13:
Hi, thanks for reporting this.

I've just worked through these various steps, the latest image that I got after rpi-update appears to be dated May 7.

I had no problem with the pyyaml installation, it installed version 5.4.1 with no trouble - I assume I've already got Ben's fix for  that... thanks Ben!

I can reproduce the jinja2 problem and have reported that to libcamera, you can track it here: https://bugs.libcamera.org/show_bug.cgi?id=33

I expect they will address the issue reasonably quickly, otherwise we can consider updating our documentation. Thanks again!

H-xie, 2021-05-14:
> Fixed missing PyYAML: https://piwheels.org/project/PyYAML/ (@H-xie please report these to piwheels in future)

Thank you for fixing, @bennuttall . And, Yes, I will do so in the future. I mistook red cross as 'cannot build'. Sorry for that.

---
Thank you for reporting the version issue in `libcamera` project. @davidplowman 

---
Thank you all for your time. It is really quick to get response.
I really appreciate you all for creating these wonderful documents. They helps a lot.

aallan, 2021-06-07:
This looks like it can be closed as resolved?

H-xie, 2021-06-07:
> This looks like it can be closed as resolved?

Yes. It’s fine to me.
Thanks again, for your replies.

closed 2021-06-07 by aallan