raspberrypi/documentation #1902: Hi, I'm afraid we've only ever run libcamera ourselves on a Pi 3 or 4, my apologies for not making this clearer on the instructions page. libcamera moves a lot of the processing load formerly on the Broadcom GPU onto the ARM cores, so it helps if they are reasonably capable. It may be that it optimises by default for a more powerful type of ARM processor than you have, so you could try deleting and remaking the build directory, and running cmake like this:

H-xie, 2021-05-14:
Hi, I'm afraid we've only ever run libcamera ourselves on a Pi 3 or 4, my apologies for not making this clearer on the instructions page. libcamera moves a lot of the processing load formerly on the Broadcom GPU onto the ARM cores, so it helps if they are reasonably capable. It may be that it optimises by default for a more powerful type of ARM processor than you have, so you could try deleting and remaking the build directory, and running cmake like this:
`cmake -DENABLE_COMPILE_FLAGS_FOR_TARGET=none ..`
Unfortunately, I would expect that you may encounter further problems down the line.

> Hi, I'm afraid we've only ever run libcamera ourselves on a Pi 3 or 4, my apologies for not making this clearer on the instructions page. libcamera moves a lot of the processing load formerly on the Broadcom GPU onto the ARM cores, so it helps if they are reasonably capable. It may be that it optimises by default for a more powerful type of ARM processor than you have, so you could try deleting and remaking the build directory, and running cmake like this:
> `cmake -DENABLE_COMPILE_FLAGS_FOR_TARGET=none ..`
> Unfortunately, I would expect that you may encounter further problems down the line.

 `cmake -DENABLE_COMPILE_FLAGS_FOR_TARGET=none ..`
Could this information be added to the document? 
Maybe [this page](https://www.raspberrypi.org/documentation/linux/software/libcamera/README.md

_Originally posted by @davidplowman in https://github.com/raspberrypi/libcamera-apps/issues/31#issuecomment-839667790_

lurch, 2021-05-14:
If it affects the Pi Zero, I suspect it'll also affect the Pi 1 too, so we need someone to experiment and test where the cut-off lies, e.g. does the Pi2 also need `cmake -DENABLE_COMPILE_FLAGS_FOR_TARGET=none ..` ? :shrug: 
(unfortunately I don't have the time for this myself)

davidplowman, 2021-05-14:
I probably need to look up exactly how I tell the different kinds of Pi apart. The build script is for Cortex A-53 (or later) systems, but I think that's a better fix than changing the documentation.

davidplowman, 2021-05-17:
That fix has been merged now and should resolve the problem. We should be able to close this issue now, though please advise if there are still problems.

lurch, 2021-05-17:
Thanks @davidplowman , closing this issue as requested.

@H-xie Please let us know if you're still having problems with the latest version of libcamera.

closed 2021-05-17 by lurch

H-xie, 2021-05-18:
Thank you for the responses.
No further problems for now.

raffaeler, 2022-06-17:
@davidplowman I just installed bullseye (and fully updated) on a RPi zero W2 and `libcamera-hello` fails with a `failed to import fd 19`.
Is the RPi zero W2 supposed to work with `libcamera`?

davidplowman, 2022-06-17:
Hi, this is a closed documentation issue, so it might be better to file a bug report on the [libcamera-apps repo](https://github.com/raspberrypi/libcamera-apps). Before doing that, maybe just double-check the [documentation](https://www.raspberrypi.com/documentation/accessories/camera.html#preview-window). Thanks very much!

lurch, 2022-06-17:
"Raspberry Pi Zero" by itself always refers to the "Zero 1".
Our [official naming](https://www.raspberrypi.com/products/) is "Raspberry Pi Zero" and "Raspberry Pi Zero 2 W". In software-terms, I believe the Zero 2 should behave like a Pi 3, just with less RAM.