raspberrypi/documentation #1903: [bcm2711 peripherals] No explanation for PACTL_CS?

DenizUgur, 2021-05-14:
Hello,

I'm not sure if this is forgotten or I couldn't find any explanation but on the page 62, there is a line:

> The alternate DREQs are available by changing the DMA_CNTRL_MUX bits in the PACTL_CS register in the peri audio control block.

Which I believe it's necessary to enable DMA for SPI4 and SPI5. Unfortunately, I couldn't find the specific format for PACTL_CS. So, I'm not quite sure how should I interpret this quote. Any explanation would be great.

Thanks

lurch, 2021-05-14:
Correct, this is currently undocumented. I believe it _might_ be a similar situation to https://github.com/raspberrypi/documentation/issues/1209 where "We've published all we can at this time." ? :shrug: 
ping @pelwell

DenizUgur, 2021-05-14:
Oh, I was afraid of this response 😞 Well for the time being may I request an explanation to enable DMA for SPI4 and SPI5?

DenizUgur, 2021-05-16:
Hello, kindly bumping this issue. Are there anyone that may assist me with this situation?

pelwell, 2021-05-16:
At the weekend?

DenizUgur, 2021-05-17:
Haha, sorry. Been working on this for a quite some time now... I guess I lost the track of time 😀

pelwell, 2021-05-17:
The BCM2711 register PACTL_CS is at hardware address 0xfe204e00 in ARM space (0x7e204e00 in VPU space). There are three DREQ selector bits:

Bit Field | Flags | Field Name | Description
-- | -- | -- | --
25 |   | DMA_CNTRL_MUX_1 | DMA This controls the source of DMA DREQ channel 21 and 22: 0 selects UART5 TX on channel 21 and RX on channel 22,1 selects SPI5 TX on channel 21 and RX on channel 22
24 |   | DMA_CNTRL_MUX_0 | DMA This controls the source of DMA DREQ channel 19 and 20: 0 selects UART3 TX on channel 19 and RX on channel 20, 1 selects SPI4 TX on channel 19 and RX on channel 20
23 |   | DMA_CNTRL_MUX_2 | DMA This controls the source of DMA DREQ channel 1: 0 selects DSI1, 1 selects PWM2

Other bits should be treated as reserved - in this case you can either write as zero or write back the previously read value, just don't go setting bits that weren't already set. 

There is no official support for changing the DREQ selectors - the firmware configures DMA_CNTRL_MUX_2 for PWM audio, and the other two are left with the reset value of 0. I suggest you use `sudo busybox devmem` to set the values you need, i.e. to use the SPI4 and SPI5 DREQs without breaking the PWM audio:
```
$ sudo busybox devmem 0xfe204e00 32 0x03800000
```

@lurch This information can be added to the documentation.

DenizUgur, 2021-05-17:
Well that worked flawlessly. Thank you for the comprehensive answer.

Enjoy your week 😉

lurch, 2021-05-17:
I'm going to re-open this, until after I've added @pelwell 's extra info to the documentation :slightly_smiling_face:

butrus, 2022-01-02:
> busybox devmem 0xfe204e00 32 0x03800000

I'm sorry to ask here but didn't found a better place: Is there a way to do this from a .dts (device-tree) file?

pelwell, 2022-01-04:
No.

lurch, 2022-01-18:
> I'm going to re-open this, until after I've added @pelwell 's extra info to the documentation

This has finally been done! :tada: 
https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf

closed 2022-01-18 by lurch