raspberrypi/documentation #1915: Schematics "USB Power In" shows incorrect CC1 and CC2 pulldown

RowanP4069, 2021-06-17:
The [schematics](https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_4b_4p0_reduced.pdf)  (accessed 2021-06-17) have not been updated for the [new ](https://hackaday.com/2020/02/23/raspberry-pi-slips-out-new-pcb-version-with-usb-c-power-fix/) _USB-C Power In_ circuit. They still show CC1 and CC2 shorted and pulled down by a single 5K1 resistor (R79).

lurch, 2021-06-17:
ping @jimbojr / @mike-stimson

aallan, 2021-06-17:
Schematics will be moved outside of the documentation repo during the reorganisation, see #1911. We may look at this then.

seamusdemora, 2021-06-20:
After reading #1911, I don't understand... The schematics aren't a *3rd party product* AFAICT - are they? Also - open-sourcing the schematic would presumably resolve the workload issue, and be a significant aid to the user community.

JamesH65, 2021-06-20:
The current schematics will be available  in the new docs world, just in a different place. Updating them is a different issue and may or may not happen.

seamusdemora, 2021-06-20:
Right... when I said "I don't understand", I meant that I don't see the point of having schematics if they're not going to be kept up to date.

aallan, 2021-06-20:
> I meant that I don't see the point of having schematics if they're not going to be kept up to date.

We have a very limited amount of engineering time. It's on a list, but the list is very long, and it's not near the top. I'll be looking at this issue, alongside other similar issues to do with ongoing maintenance of the documentation, as part of #1911.

seamusdemora, 2021-06-20:
> We have a very limited amount of engineering time. It's on a list, but the list is very long

Yes, I understand that there may be a resource issue, and prioritization is necessary. But you didn't remark on the *open-sourcing* alternative. It's obvious that there **is** an up-to-date schematic somewhere, and it's obvious that it must be maintained. Two questions arise here: 

   1. Would it not be more *resource-efficient* to make the up-to-date/complete schematic available - rather than spending additional resources to create and update a **partial** schematic? 
   2.  Is there a **policy** within the Foundation that prohibits open-sourcing a complete & up-to-date schematic? 

From my own experience, and many comments I've read over the past couple of years, my *sense* is that the limited schematic is not very useful. It would follow that an incorrect/out-of-date schematic would be even less useful. 

And so - given all the above, it would seem that a more *forthcoming* statement would be a preferable closure to this issue.

JamesH65, 2021-06-20:
Raspberry Pi's are not open source hardware, and we are unlikely to change that stance.

hachi, 2022-01-25:
Can we bump this open again please... the docs site update happened some months ago and this wasn't addressed and now we have no external tracking for when this gets done.

JamesH65, 2022-01-25:
There is no schedule for this, nor is there any guarantee it will actually be done.

OliverZuckerberg, 2023-03-08:
Sad.

ryao, 2023-06-08:
> Raspberry Pi's are not open source hardware, and we are unlikely to change that stance.

Then what is this:

https://datasheets.raspberrypi.com/rpi4/raspberry-pi-4-reduced-schematics.pdf

They published a schematic. They should update it to reflect the state of the production hardware. This can be useful to people developing hardware based on the CM4.

TamtamHero, 2023-09-12:
>    Raspberry Pi's are not open source hardware, and we are unlikely to change that stance.

Pretty disappointed by this answer. It's absurd to publish the v1 schematics but not update them later. What are you guys afraid of ?

I'm currently learning about electronics and USB-C. I was about to make the same mistake you did, connecting CC1 and CC2 together, but I've been lucky enough to come across the RPI 4 case.  "Great!" I thought. "I'm gonna look at their fixed schematics to see how they fixed that!"
And then I discover that it's not fixed yet after many years, for dumb pretexts.
I've been advocating for Raspberry boards since the first one, but I see that I've been misled. Sad.

aallan, 2023-09-12:
As I said up thread. We have a small engineering and documentation team. It's on a list of things that need doing, but the list is very long, and it's not anywhere near the top. Calling us names isn't going to help your case.

seamusdemora, 2023-09-12:
@aallan said, "Calling us names isn't going to help your case."

Nor does it seem it's going to hurt the case either. :)

closed 2023-10-10 by aallan