raspberrypi/documentation #1931: Missing LED flash code

Thermi, 2021-06-28:
0 long flashes, 2 short flashes is not explained

[Doc ](https://github.com/raspberrypi/documentation/blob/master/configuration/led_blink_warnings.md)

JamesH65, 2021-06-28:
This is not a flash code that is output by the bootloader, so I suspect you have hardware that is unable to even start up correctly. The LED flash code will repeat, and I suspect this is a one off thing?

Is this a new Pi, or has it suddenly started failing?

Thermi, 2021-06-28:
This is with ArchLinux ARM 64bit ARMv8 latest. It recently started flashing, since I reflashed the SD card. I have not yet had the chance to check if a different SD card works. At the very least the journal is not being written to, so it seems the OS can not write to the SD card or does not get as far as starting journald.

I don't have any means of attaching a display right now, so network diagnosis and flash codes is what I got.

JamesH65, 2021-06-28:
I doubt a change of SD card or OS will help - it's appears the Pi is broken, since that LED flash sequence is not one output by the bootloader, so is more than likely random firing of the LEDs as the device attempted to power up.

I presume the pattern just happens once and does not repeat?

JamesH65, 2021-06-28:
(BTW, this is the docs repo, this should really be handled on the forum)

Thermi, 2021-06-28:
Yes, I can figure this all out myself just fine. This is just the report that this flash code is missing. :P

aallan, 2021-06-28:
I'm not entirely convinced it's an intentional reporting code, it's not one output by the bootloader. I'll check internally and update the table if anyone can think of anything. Closing.,

closed 2021-06-28 by aallan

aallan, 2021-06-29:
Response from our engineering team was that, "_It’s not an error code and all fatal errors repeat (forever). The bootloader switches on the green led at power on but since it’s muxed with the bootloader spi flash chip select gpio there’s some unavoidable flicker as the ROM and second stage read from spi flash. Best guess is that the ROM attempted to read from flash and it subsequently died for reasons unknown eg broken pmic, shorted GPIOs etc_"