raspberrypi/documentation #1934: Setting up a Raspberry Pi as a bridged wireless access point

gugabguerra, 2021-07-01:
[documentation/configuration/wireless/access-point-bridged.md](https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point-bridged.md)

I got it working easy following this document step by step, with just a simple additional command:

- You must enable IPv4 Forwarding to allow clients to reach the DHCP server

`sudo nano /etc/sysctl.conf`
add or uncomment: `net.ipv4.ip_forward = 1`

then...
`sudo systemctl reboot`

aallan, 2021-07-01:
There is an open PR for this bit of the documentation, see #1929. Could you either add it to that PR, or open a new one. Thanks!

gugabguerra, 2021-07-02:
Thanks. Sorry if I posted into the wrong section

closed 2021-07-02 by gugabguerra