raspberrypi/documentation #1935: Documenting GPIO 3's "wake from HALT" function

Gadgetoid, 2021-07-01:
As raised in https://github.com/Gadgetoid/Pinout.xyz/pull/413 there doesn't appear to be any primary source of documentation for the "wake" functionality exposed by GPIO 3 (Physical pin 5).

Some scant, out of date documentation is available here - https://elinux.org/RPI_safe_mode#cite_note-1

And there are a scattering of other references, forum posts and mentions- many lamenting the lack of documentation, but nothing canonical.

1. Would this be an appropriate addition this documentation?
2. Where would it fit?

Points we should probably cover:

1. Works only in HALT state when the Pi still has power
2. GPIO 3 was chosen due to the external, strong pull-up that keeps the pin in a known state and avoids accidental wakes
3. Pin cannot be changed
4. Should be shorted to ground via a button to trigger a wake (physical pin 6 is *right there*)
5. Which versions of the Pi is this supported on? (All? or post a given firmware version?)

aallan, 2021-07-01:
> 1. Would this be an appropriate addition this documentation?

Yes

> 2. Where would it fit?

Bearing in mind the looming nature of #1911…

Potentially as a section in the [GPIO](https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/gpio) documentation?

lurch, 2021-07-01:
Or possibly somewhere in https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/bootmodes ? :shrug:

aallan, 2021-07-01:
> Or possibly somewhere in https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/bootmodes ? 🤷

Jinx! I just edited my comment to add that one. 😁

aallan, 2021-07-05:
I'm probably going to be freezing the current markdown-based documentation tree, and deleting any pending PRs, at the end of the week (see #1911 for more information as to why). It probably makes sense to pause this until after the new documentation site is released and then PR ontop of the new Asciidoc documentation (end of the month maybe?)

Gadgetoid, 2021-07-05:
I’d figured as much. I’d need to get a feel for the new style/format and figure out where to put this. Plus get some authoritative answers about the what’s, why’s and where’s of it all 🤣

aallan, 2021-08-09:
New site is live! 😄

Gadgetoid, 2021-08-09:
It's... uh... nice n'all... but a little bit completely broken without Javascript... and somewhat difficult to browse when it runs on continuously with no pagebreaks and very little differentiation between first and second level headers :grimacing:

closed 2021-10-24 by github-actions[bot]