raspberrypi/documentation #1938: config.txt section paragraph in "Raspberry Pi 4 bootloader configuration"

seamusdemora, 2021-07-06:
As I read the paragraph under the `config.txt section` of the [Raspberry Pi 4 bootloader configuration](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md), I struggle for clarity. Here's what it says to me: 

>If one adds a a section called `[config.txt]` to the bootloader configuration (e.g. using `sudo -E rpi-eeprom-config --edit`), then that section is appended to `/boot/config.txt` 

It goes on to say:

>This can be used to automatically apply settings to every operating system, for example, dtoverlays. 

Does this also mean that I could add (for example) the parameters `WAKE_ON_GPIO=0` and `POWER_OFF_ON_HALT=1` directly to my `/boot/config.txt` file, and avoid the editing step `sudo -E rpi-eeprom-config --edit`?? 

In other words - rather than adding a section called `[config.txt]` to the bootloader configuration, could I simply add these parameters to `/boot/config.txt`, and achieve the same effect?

lurch, 2021-07-06:
Short answer: No :wink: 

Long answer: the [eeprom config](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md) (stored inside the eeprom, and edited with `sudo -E rpi-eeprom-config --edit`) and [`/boot/config.txt`](https://www.raspberrypi.org/documentation/configuration/config-txt/README.md) (stored on the first FAT partition of the SD card, and edited with `sudo nano /boot/config.txt`) are very different things. Have a read of https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow_2711.md for the full details of the Pi4 bootup process.
The `WAKE_ON_GPIO` and `POWER_OFF_ON_HALT` are settings used by the bootloader eeprom code, and this code is running before the SD card has even been mounted yet (which is why those settings **can't** be stored in `/boot/config.txt`).
This is only an educated guess, but I'm guessing that the `/boot/start.elf` firmware (that runs from the SD card at a later stage in the boot process) reads `/boot/config.txt` from the SD card and _also_ reads the eeprom-config and appends the `[config.txt]` section of the eeprom-config to the values read from  `/boot/config.txt`.

seamusdemora, 2021-07-07:
OK, that's a good and reasonable explanation, but do you feel the `config.txt section` is clearly written?...  

For me, reading the bootflow details does not make this any clearer. Even knowing the eeprom code was running first doesn't mean that start.elf couldn't effectively update the configuration after reading /boot/config.txt - at least not to me it doesn't. And I'll guess that the documentation in `config.txt section` wasn't written for those who are knowledgeable of such details. :) 

Now, with the benefit of your explanation, I'll guess that `config.txt section` is saying the converse of my question - in other words, this feature allows one to put configuration instructions in the eeprom config instead of `/boot/config.txt`.  Is that the correct interpretation?

lurch, 2021-07-07:
Don't forget that the eeprom-config also contains things like `BOOT_ORDER`, which tells the code-running-from-eeprom where it should even _try_ looking for `start.elf` & `config.txt` :wink: (which is why this kind of thing has to be done way before `start.elf` is even loaded)

> Now, with the benefit of your explanation, I'll guess that config.txt section is saying the converse of my question - in other words, this feature allows one to put configuration instructions in the eeprom config instead of /boot/config.txt. Is that the correct interpretation?

I believe so, but @timg236 is the person who knows how all this works... :slightly_smiling_face:

seamusdemora, 2021-07-07:
I can run an experiment to verify this. I hoped that I'd be able to write those EEPROM config variables to `/boot/config`, but not to worry - I got what I needed, and thanks to you for that.

lurch, 2021-07-10:
Is anything else needed here @seamusdemora , or can this be closed?

seamusdemora, 2021-07-10:
AFAIC, it can be closed - I'll do that now. That's not to say I've changed my view that the current wording is not a model of clarity, but I suppose anyone else that finds it confusing can ask another question.

closed 2021-07-10 by seamusdemora

lurch, 2021-07-10:
We're going to be locking this repo soon (see #1911 ), but feel free to open a PR when we open it back up again.